Ultimaker/Cura #3173: problem with cura 2.3.1

buckles94, 2018-01-24:
Version: 2.3.1
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
  File "C:\Python34\lib\codecs.py", line 319, in decode
MemoryError

ChrisTerBeke, 2018-01-24:
2.3.1 is an old version of Cura. Can you try the newest version?

fieldOfView, 2018-01-24:
@buckles94 is probably using a 32 bit version of Windows.

Press winkey + r, and type %LOCALAPPDATA%\cura followed by enter.
In the folder that opens, remove cura.log
Cura should now launch normally again.

ChrisTerBeke, 2018-01-24:
Ah, that's true. 2.3.1 is indeed the latest 32 bit supporting version. Like @fieldOfView indicates, the log file might have become too big. Removing it as explained by him will fix that.

buckles94, 2018-01-24:
yep spot on. thanks very much

closed 2018-01-24 by buckles94