Ultimaker/Cura #3174: [UX] Also add the supports/brims/rafts/skirts in solid model view

MathijsG, 2018-01-24:
I often get confused why I don't see my setup changes or settings about the supports or the brims/rafts/skirts, and then after a while I discover the view was in 'solid' mode, instead of layer mode.

However if you ask me, it would be more logical that at least some of these things were also visible on the solid model setting (which is the default visual setting).

fieldOfView, 2018-01-24:
The problem is that supports and adhesion are created during slicing, and exist only as layers (not as a mesh).

jackha, 2018-01-25:
That's why I use layer view so often ^^

MathijsG, 2018-01-25:
If this is technically not possible, maybe we should consider some notification or something that these things will only be visible in layer view? For some people this won't that clear the way it is currently :)

Ghostkeeper, 2018-02-13:
I'd like to try to prevent textual "warnings" like that as much as possible, especially if it's in the happy flow, like it is here. You've just loaded a model. You don't want to see warnings that it might not look like the actual print there. The view is intended to show the source model, and it should be clear as such.

We could consider automatically switching to layer view once slicing is complete. But we'd have to consider that you're doing stuff automatically then which takes control out of the user's hands. Especially when positioning a small model that could be annoying.

MathijsG, 2018-02-13:
I am with you on automatic switching to layer mode, this is how S3D deals with this too. There are many starters in this hobby scene, and expecting that everyone knows there are different representations of a model, some with the supports visible and some where they aren't, is a big assumption if you ask me.

And I disagree with your 'happy flow', people won't be happy if they print something with supports because it was still on, but didn't show up on the model preview. Not every notification has to be a red blinking error with loud sounds. But now the user isn't notified at all and being left in confusion, all to please the 'happy flow'.

ianpaschal, 2018-02-13:
It seems like a sort of strange case IMO. @Ghostkeeper is right: warnings should be kept to a minimum. Even if they're small, they're still noise. Warnings are absolutely not improving the UX in my opinion.

It could be true to say that the mesh view doesn't do enough to distinguish itself from layer view but then then again... one looks like solid mesh and one looks like grey ghosts or stacked spaghetti. To be honest, I don't really understand how you keep forgetting what you're looking at.

MathijsG, 2018-02-13:
If you forget how complicated 3D printing for a beginner can be it's very easy to not understand the point, if you ask me. How does one know the visual they see is 'wrong' and every setting they change don't affect the visual model at all? How does one know of mesh view at all? The program just starts itself with the regular clean non-sliced view, imagine you're new to the printing world, maybe people don't even know you're supposed to see a more sliced model. They only encounter that changes they make don't reflect back on the model thee see on screen.

Not every user notification or mention is a 'warning', I suggest you people read a few UX books, because feedback to the user is a big topic in these. And now the feedback you get in this program is quite minimal, rendering people in confusion.

ianpaschal, 2018-02-13:
Your hypothetical example (again, I'd be really curious to know more specifics) is hypothetically true. But for any hypothetical example, there exists a hypothetical counter example. I could just as well argue that someone who is really unfamiliar with slicing (and apparently 3D printing and Google) might be really thrown off by importing their model and immediately being confronted with all kinds of extra geometry... weird lines around the base and places that look like extra faces are being inserted between overhangs and the base, etc. After all, in this hypothetical, they don't understand what slicing is, why would they understand the magic appearance of a brim or supports?

And again, the different views are always accessible, in plain sight, and use different graphic styles to indicate what they are.

Now if you want to get specific like, "The grey used in solid view looks too much like the grey ghost in layer view," that's a fair point. They're obviously both intended to be neutral but perhaps some other style is possible like x-ray.

But in this case, please be specific about what you feel is confusing instead of "It's confusing and maybe the user knows nothing."

> I suggest you people read a few UX books, because feedback to the user is a big topic in these.

I have a MSc in Design for Interaction, as do others in the Cura team. Please don't be patronizing.

Ghostkeeper, 2018-03-07:
Maybe we could leverage a geometry shader to "drop down" all triangles that are rendered red to the build plate and then render that transparently in some way. But indeed we'd have to be careful because the "Solid view" really is intended to show the input model. Like Ianpaschal said, it could be confusing for a novice to see extra geometry there that wasn't in the input model so it must be made clear which parts of the model are original and which are generated.

Other stuff like brim and raft requires polygon offsets of the initial layer which is quite difficult to do before slicing. You'd need to know the shape of the first layer, which is a slice. And the offset itself is also not trivial.