Ultimaker/Cura #3177: Cura 3.1 and 3.2beta doesn't work with old profiles/machines

printingotb, 2018-01-24:
**Application Version**
Cura 3.1 and 3.2Beta

**Platform**
Win7

**Steps to Reproduce**
First I had the same problem as described here but I got past it by emptying the appdata local and roaming Cura folders: https://github.com/Ultimaker/Cura/issues/2837
I then recreated my machine profiles, one CR10 and one custom FDM printer. So far everything is good.
The problem appears when I want to import my profiles that I had previously exported. Only 7 out of 50 profiles was able to be imported again (no error message). If I remove the custom fdm printer, I can import 43 profiles.
If I add another type of machine and remove the CR10 and the custom one. I get a message "... profile doesn't match your current machine  ,could not import it."

By the way, it would be easier to make backups of my profiles if I didn't need to export them one at a time and also have to manually type in their names.

**Expected results**
How do I get all my profiles to work with the newer Cura versions?

**Additional Information**
Please let me know if you need additional info from me.

thetobby, 2018-01-24:
Can you upload one example of the profiles that fails to load?

printingotb, 2018-01-24:
Sure!
[WDI3 PLA.zip](https://github.com/Ultimaker/Cura/files/1661520/WDI3.PLA.zip)
Hmm when I had only the CR-10 machine, I could load one more than the 43 successful profiles but when I try another one, it says "Successfully loaded" but it doesn't load. Sorry if this is confusing but it may provide a clue.
Anyway, attached is a profile that doesn't load when I have both printers setup.

BenMarolt, 2018-01-25:
Anyone else thinking export/import buttons for printers would be nice?

ctzsnooze, 2018-01-25:
Yep +1 for import/export printers

jackha, 2018-01-25:
@BenMarolt if you save an empty project file, it's already very similar as exporting a printer, including all settings and profiles. When you load the project file from a different Cura, the printer is created on that Cura.

printingotb, 2018-01-25:
Testing some things.
If I try to import the same Profile several times, I get the message "Successfully imported profile "CR10 PLA #n" where "n" increases for each attempt. The profiles doesn't show up but Cura recognizes them as duplicates.
Is it possible to look in the Profile and see what kind of machine it expects?

When I save an Empty project file in Cura 3.0.4 and try to open it in Cura 3.2beta, I get error: Invalid file - Failed to load. As would be suspected, it opens in 3.0.4
[Empty projekt.curaproject.zip](https://github.com/Ultimaker/Cura/files/1663506/Empty.projekt.curaproject.zip)

printingotb, 2018-01-25:
Some more clues..

I did a new install and cleaned the appdata again leaving the folders in place.
Now I can suddenly import the same project file saved with the Cr10 machine active.
However, I cannot import a project file saved with the Custom FDM machine afterwards.

If I again make a new cleaned install I can suddenly import the project file saved with the Custom FDM. However, the settings profile that came with that project is active but doesn't show up in the list under the protected profiles.
If I make a change to the profile, I can now use the dropdown to create a new profile from current settings and this profile is actually there! (Yes this was one of the 43 profiles that did not work before).

printingotb, 2018-01-25:
I just realized. Of the 50 profiles I have, 43 work with the CR10 machine. And 7 work with the Custom FDM machine. Those 7 are the ones that doesn't work with the CR10 and vice versa. How could I miss that?

When creating a settings profile, should it be locked to a certain machine? 
In 3.0.4 and before, all my profiles are visible in both my machines.
As it is now, the profiles that are visible on each machine has nothing to do with what machine they are made for.

BenMarolt, 2018-01-25:
@jackha Cool, I never realized the printer would be created from the profile. I've always tried to get the printer to show up first before importing my profiles. I'll keep this in mind next time I install.

jackha, 2018-01-29:
@printingotb Sometimes the configuration folder become incorrect, that can explain why the empty project file didn't load first and then it does after cleaning the preferences folder.

If it doesn't work, can you post the log files also so we can have a look? 

Good point, profiles that are for one machine only. We currently don't support setting compatibility for different machines in the UI, the only way to do that now is to edit the files manually.

printingotb, 2018-01-29:
Ouch, the log file is 395MB. Seems to be all versions logs, not only the latest?

printingotb, 2018-01-29:
[Cura Log cut.txt](https://github.com/Ultimaker/Cura/files/1672926/Cura.Log.cut.txt)
I cut out a piece at the end beginning at 2018-01-22.

jackha, 2018-01-29:
Thanks, we'll have a look.

jackha, 2018-01-29:
Can you reproduce this problem from a clean install? Would be very helpful.

printingotb, 2018-01-29:
Like this? (If so I have been doing it 3 or 4 times already).
Uninstall Cura 3.2 beta
CCleaner
Install Cura 3.2 beta
Remove content of appdata local and roaming foldes but leave folders in place.

jackha, 2018-01-29:
Oh sorry, I meant do you get the problem if you do that? It looks what you did solved the problem ^^. I'm trying to get us to reproduce it so it would be easier to debug instead of only looking at the logs and "guess" what happened.

printingotb, 2018-01-29:
Yes, I get the problem with a clean install :)
Let me know if I should do anything else.
Would love to be able to use the never versions with my old profiles. I am dependant on them for my work. I suppose it would be possible to recreate them manually but it's 50 profiles!

jackha, 2018-01-29:
By "Remove content of appdata local and roaming foldes but leave folders in place." did you mean the specific 3.2 folder(s) or the generic cura folder?

printingotb, 2018-01-29:
The specific 3.2 folders.

jackha, 2018-01-29:
That is really really strange. Just thinking out loud. Keeping the 3.2 folders but not their contents will not upgrade some old Cura versions. So at first startup you also get the add printer thing. Ehm so how did you solve it? Did you solve it or do you have a workaround?

What happens if you open and close Cura after clean installing? Does it work then?

printingotb, 2018-01-29:
First, I had to empty the 3.2 folders to be able to start Cura or I would get the error explained in this issue: https://github.com/Ultimaker/Cura/issues/3170
Now that Cura starts, I have to create new machines through the dialog.
My 3.0.4 profiles doesn't work with the new machine profiles. Cura thinks they are there but they do not show up in the list and I cannot choose them.
Restarting Cura doesn't help.

Also, at some point I emptied the folders of my latest working version 3.0.4 and made new machines and imported all my profiles. They work in that version. This action made it possible to run the new Cura 3.2 without cleaning the 3.2 folders before starting. (Just noticed).

It is likely that most of my profiles stem from a profile that I created in some of the earlier Cura 2.0 versions. Maybe something with my settings profiles are no longer compatible with the new machines?

printingotb, 2018-01-29:
Restarting Computer doesn't help either.
Adding the two first printers from the other list gives the same result as with my CR10 profile. 43 of my profiles work. My custom FDM printer machine (for wanhao di3) only accepts 7 profiles. This is the machine that must be able to use all my profiles.
Adding an Ultimaker 3 however, none of my profiles work (wrong nozzle diameter?)
Should I try with any another machine profiles?

jackha, 2018-01-29:
Thanks. We'll have a look.
devs: CURA-4885

davidwyl, 2018-02-04:
current;y im at 3.0.4, i unable to upgrade to 3.1 (it will crash before it start), today i try 3.2 beta im getting the same python error...
i deleted all the 3.2 data in Local & roaming, 3.2 run without problem but it start fresh without my printers & profiles.
Is there a way to install 3.2 beta without loosing all my profiles and printers?

printingotb, 2018-02-04:
@davidwyl Have you tried exporting profiles from Cura 3.0.4?
You should be able to save an empty project in 3.0.4 and open that in 3.2 and the active machine should transfer.

davidwyl, 2018-02-05:
@printingotb thanks. Save empty project and import works...

printingotb, 2018-02-07:
I installed 3.2 today and now I can actually import settings profiles. They are still given a "#2" as if the profile was already there. Before, they didn't show up in the list.
However, if I import all profiles with Machine1 active and then change machine, the profiles disappear. If I import them again with Machine2 active, they get the name "Profilename #3". 
The same happens for newly created profiles.

Is a profile I create for Machine1 not supposed to be available to Machine2?

printingotb, 2018-02-08:
Machine1 is a Custom FDM printer and Machine2 is CR10 from the list.

jackha, 2018-02-08:
It's intended that profiles are machine specific. There's also a check for incompatibility and apparently those two machines are compatible. I'm not quite sure about the naming, I think it just avoids duplicate names.

printingotb, 2018-02-08:
When did they start to be machine specific? My current cura version 3.0.4 doesn't care what machine I have selected, all my profiles are always visible.

jackha, 2018-02-08:
Yeah I think that was actually a bug ^^. Like tweaks for one machine does not necessarily work on a different machine.

closed 2018-03-27 by ianpaschal