Ultimaker/Cura #3178: Small gear icon next to category name to change visibility in settings is gone

mpsdskd, 2018-01-24:
I'm missing the small gear icon next to the category name to change settings visibility.
Is that gone intentionally or is that a bug?

[image]

I know that I can right-click to change settings visibility, but I liked that gear icon feature.

OT:
Big shout-out to all developers, 3.2 got a lot faster on my weak device.

I had problems with my profiles again, but since #2912 I know how to work around this.

fieldOfView, 2018-01-24:
Does not reproduce for me (ie: the gear icon shows up just fine running the beta on Windows).

Something else that caught my eye in your small screenshot is that it does not show a blue underline under the category. So the real issue seems to be that somehow mouse hover is not being detected.

mpsdskd, 2018-01-25:
I can confirm that no blue line appears under the category name on mouseover.

However tooltips for each item in these categorys are shown on mouseover and input fields light up blue on mouseover.

Ghostkeeper, 2018-02-13:
Also doesn't reproduce for me, using the 3.2 stable release on Ubuntu 17.10.
[image]

ianpaschal, 2018-03-09:
Also doesn't reproduce for me. We're going to need some logs to look for QML bugs or something. @mpsdskd could you share your log file please?

mpsdskd, 2018-03-12:
Of course I can:
[cura.log](https://github.com/Ultimaker/Cura/files/1801802/cura.log)
I guess this is the one, you are looking for?

Ghostkeeper, 2018-04-05:
That's the log file, yeah. But the log file doesn't seem to contain any indication of anything out of the ordinary. Maybe only "Object destroyed upon incubation" is something but that's not something we can trace down from the log :crying_cat_face:

ChrisTerBeke, 2018-05-23:
No new info and can't reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke