Ultimaker/Cura #3179: Cura 3.2 Beta does not start; mac

filipgoc, 2018-01-24:
**Application Version**

3.2 Beta

**Platform**

Mac 10.12.6 Sierra

**Qt**

**PyQt**

**Display Driver**

AMD Radeon R9 M370X 2048 MB
Intel Iris Pro 1536 MB

**Steps to Reproduce**

Try to open Cura. 

**Actual Results**

The Cura 3.2 beta splash screen shows, but then disappears after a while and that's it.

**Expected results**

**Additional Information**

I have tried restarting the system, no luck. 

Here's the log from 3.2 folder, abridged to the relevant date. Fingers crossed. 

[cura.log.txt](https://github.com/Ultimaker/Cura/files/1661645/cura.log.txt)

ChrisTerBeke, 2018-01-24:
Could you share a zipped version of your user data folder? Somehow there are files in there that can't be parsed properly, but we can't reproduce on our computers.

ChrisTerBeke, 2018-01-24:
Devs: CURA-4863 and CURA-4864.

filipgoc, 2018-01-24:
@ChrisTerBeke do you need everything in the Cura support folder or can you be more specific? The whole thing with all the versions is huge -  several gigabytes.

ChrisTerBeke, 2018-01-24:
Only the 3.2 version folder. We then install that on our own machine and try to reproduce the exact error for debugging purposes.

filipgoc, 2018-01-24:
Knock yourselves out 👍 https://we.tl/KvUrKWOwHT

LipuFei, 2018-01-29:
Hi @filipgoc , I managed to fix your cura folder. You can download it here: https://send.firefox.com/download/18aa2fdda2/#qeEjfGufeBKy3QUh4S5JYg

There are several issues. One of them is caused by the name change from `jellybox` to `imade3d_jellybox` (see commit: https://github.com/Ultimaker/Cura/commit/0d4a74182b3a640ea02cbdc0ebd7db76b35f1ef1#diff-d18710f89128620a7719d8783cea4181).

filipgoc, 2018-01-29:
@LipuFei THanks! Nice going there!

What did you do to 'fix the folder? Maybe let's start with that. 
As something for sure has changed in 3.2 upgrade procedure, or by the nature of what you did. 

Cura 3.1, 3.0... all of those simply did not load old 'jellybox' machine profiles. Just dropped it. Cura 3.2 as you fixed it now does load them (yay), but that's of limited use since he 'jellybox' machine does not exist anymore. It's not supported anymore. I was fine with that behavior (though a popup notification would be a nice touch since it's user-facing...). 

But now it didn't load at all, and, after your magic, it loads even abandoned machine instances. My question is what does this mean for the users upgrading? Do we have to instruct people to delete old instances of things? 

What you did here is a **Better solution** I think as it allows users to manage/delete/update their old printer instances themselves. At least they have the choice. 

But I also think it's OK to treat machines with different IDs as different machines. Not sure that JellyBOX to IMADE3D JellyBOX should be treated as a rename (even though I called it that in the commit). Maybe there's a new machine, and another machine gets dropped. 

---

The overall question is, what do I tell people that experience the same issue as me ? :-) 

Thanks for you help!

filipgoc, 2018-01-29:
PS: a follow up question is whether it's the user folder that needs fixing, or the Cura upgrade procedure that needs fixing.

filipgoc, 2018-01-29:
PPS: I verify that I CAN run Cura 3.2 Beta with your fixes. 🎊

alekseisasin, 2018-01-29:
If this issue is solved then we can close it

fieldOfView, 2018-01-29:
@filipgoc: you made a PR that renames a file. But your configuration still points to the old filename/id. This means your configuration (and that of other people using the jellybox printer) is broken in 3.2.

filipgoc, 2018-01-29:
@fieldOfView I don't think I understand the issue. 

The old definition was filename `jellybox` with id `jellybox`.
The new definition was filename `imade3d_jellybox` with id `imade3d_jellybox`. 

Even though I said 'rename' in commit message (which is is from UX perspective), I thought the system should treat this as deleting the old one and adding a new one. 

I don't understand how the new filename/id "points" to the old filename/id.

**As for resolution:** 

The issue is resolved on my computer, but I don't know what to do when others come up with the same issue. I don't know what fixing the root cause would be. It's quite likely I made some mistake, but I'm not sure what I am supposed to fix.

filipgoc, 2018-01-29:
I'll look around more. 

What I don't get is what changed now, and what that means. 
The 'rename' PR was in March last year, and it has not cause any issues until now.

Ghostkeeper, 2018-02-14:
I don't know what changed recently then, but what fieldOfView meant was that while the definition filename and ID was changed properly, the files in your local configuration folder were not.

filipgoc, 2018-02-14:
Thanks. Let's drop this and thanks for the help! I'll presume it was just my comp that's affected unless someone cries out back to me.

closed 2018-02-14 by filipgoc