Ultimaker/Cura #3180: 3.2 Beta Won't launch

Patrick804, 2018-01-24:
[image]

**Application Version**
<3.2 Beta>

**Platform**
<Windows 10 -->

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

ChrisRiddell, 2018-01-24:
Duplicate of #3170

Patrick804, 2018-01-24:
I got it to launch. My first time here so I neglected to read previous posts. I deleted the roaming files and it started right up.

closed 2018-01-24 by Patrick804

fieldOfView, 2018-01-25:
Deleting the config should not be necessary though, so it is still a bug. But it is a duplicate of some other reports.

jackha, 2018-01-25:
We've created an issue to fix this problem before the upcoming release. Devs: CURA-4868

LipuFei, 2018-01-29:
Hi @Patrick804 , could you upload your cura directory so we can check what happened? It's located in `C:\Users\<your-username>\AppData\Roaming\cura`. Thanks