Ultimaker/Cura #3182: Incorrect nozzle temperature issues reported in #2942 not resolved in Cura 3.2.0

Stickyfeatures, 2018-01-24:
As above, Gcode still sets by default to 210 degrees. Will this be resolved in the final release?

**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

jackha, 2018-01-25:
Can you explain a bit more?

Stickyfeatures, 2018-01-25:
Yes, in several other linked threads on last version of Cura (e.g. #2942) there's a bug report that the user submitted temperature for the nozzle was overridden on dual-nozzle machines with 210 degrees. All those threads were closed with the report that this would be fixed in the next release but I'm just reporting that this hasn't been fixed and my code still sets the temp to 210 degrees

Stickyfeatures, 2018-01-25:
Ghostkeeper wrote: "Duplicate of #1296, #2091, #2546 and #2812. Only now it also manifests on single extrusion printers because those printers now also have an extruder stack."
--

ChrisTerBeke, 2018-01-25:
Mm, I can't reproduce this in the Beta, getting `M109 S215` in the G-code when setting the initial layer temperature to 215. Can you share a project file or your user data folder with us so we can try to reproduce?

Stickyfeatures, 2018-01-26:
Hi Chris, my machine is a CTC Makerbot replica for which I've created a custom profile - I run Ghostkeepers plugin to convert gcode to X3G. 

Yesterday I had a problem with 3.2 exporting to X3G at all ("scene object has no attribute g code list") - it made an X3G file but it was 0kb so I reinstalled 3.1 and the same problem appered on that. In the end I had to remove all Cura files/folders and reinstall 3.1 from scratch, now it works again but I have the same old problem. Here's the start gcode - you can see the target temp of 195 but the code gets set to 210...

```
;FLAVOR:Makerbot
;TIME:15945
;Generated with Cura_SteamEngine 3.1.0
M190 S65
M104 S195
M109 S195
M82 ; absolute extrusion mode
Right Extruder Start Gcode

M136 ; tell the firmware we're starting a build

M73 P0 ; reset build progress

G130 X118 Y118 A118 B118 ; set vref's to default

M190 S65.0 ; set bed temp, don't wait

M104 S210 T0 ; set right nozzle temp, don't wait disable right nozzle, don't wait

M104 S0 T1 ; disable left nozzle, don't wait

T0 ; home on the right extruder
```

Suffice to say that when I installed 3.2 I tested and it was the same issue which is why I reported on here. Not dissatisfied - you've made a great software - just trying to help.

ChrisTerBeke, 2018-01-26:
I think that 2nd M104 S210 line is either in the custom start-gcode or in the X3G converter?

Stickyfeatures, 2018-01-26:
Sorry - to clarify that was the Gcode for a part.

This is the start gcode line as it is in the profile:

`M104 S{print_temperature} T0 ; set right nozzle temp, don't wait disable right nozzle, don't wait`

So, by rights the {print_temperature}  should be 195 as pre-set or that print, should it not?

Ghostkeeper, 2018-01-28:
I'll upload a fixed version of the X3GWriter tomorrow so that it'll work with 3.2 as well as 3.1. You can then use the 3.2 beta to have correct temperatures again.

Stickyfeatures, 2018-01-28:
OK perfect, thanks Ghostkeeper - was the X3G converter the problem? Any idea about ("scene object has no attribute g code list") problem - with the new X3G converter fix that problem too?

Stickyfeatures, 2018-01-30:
Ghostkeeper. Can you tell me how it is possible to update a plugin in Cura please?

kelemenlajos, 2018-02-07:
I think i found something. If you install the app, and don't change the material, only switching to custom setup, the extruder temp works as expected. But after changing Material the "Printing Temperature" value isn't used anymore.

Could someone also test this?
(I reproduced the problem on my home and work computer.)

Ghostkeeper, 2018-03-02:
X3G converter was not the problem with the incorrect temperatures, as far as I know. You should be able to test that by saving to a .gcode file and looking in there with a text editor to see what temperatures are listed there. The problem that I fixed was that the API of Cura changed and the slice result now has to be obtained from the current build plate (due to the multi-buildplate feature).

You should be able to update a plug-in in Cura by going to the Plug-in Browser (under extensions) and then clicking the update button next to the plug-in in question.

I tested Kelemenlajos' finding but couldn't reproduce that. These were my steps:
1. Completely wipe all settings.
2. Start Cura. Add a Custom FDM Printer.
3. Switch to CPE+ material.
4. Load a model and save the g-code.
5. Inspect the g-code.

I see right at the beginning the following two lines:

    M104 S260
    M109 S260

This is the temperature for CPE+, not for PLA, so it appears to be setting the temperature correctly.

ColinRedGriff, 2018-03-26:
Hi

I thought I would add some more information on this issue as I am facing it too. No matter what parameters I use in the start gcode the output is always 210 degrees. It outputs if I hard code a figure into the start code but that's not really any use for normal printing.

I've created a new custom printer from scratch and add the following start gcode to it - this was just to test to try and show what is happening when trying to set the use the extrude temperature.

This is the start code I added to my printer
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S{material_print_temperature_layer_0} ; set extruder temp
M104 S{material_print_temperature} ; use material temp
M104 S250 ;hardcode 250 degress

The temperatures shown in the material section of Cura were 235 initial and 245 normal print temperature. Screen grab attached
[image]

This is the output gcode generated from it at the start of the file 
;FLAVOR:Marlin
;TIME:12380
;Filament used: 7.31912m
;Layer height: 0.1
;Generated with Cura_SteamEngine 3.1.0
M82 ; absolute extrusion mode
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S210 ; set extruder temp
M104 S210 ; use material temp
M104 S250 ;hardcode 250 degress

I'm not sure if this a particular issue with the way that starting gcode is processed or not, further down in the file the temperature does get set to the correct value (M104 S250), but not before a significant amount of printing has been done.  I've attached the entire gcode output from my test file in the zip file if that is any help. 
[test.gcode.zip](https://github.com/Ultimaker/Cura/files/1849494/test.gcode.zip) Cura 3.1.0

fieldOfView, 2018-03-26:
@ColinRedGriff, what sort of printer do you use, and how did you add it to Cura? Is it a Custom FDM Printer, or another type?

Could you try adding your printer to Cura again using the Add Printer dialog in the Printers pane of the preferences, and see if this still reproduces on the second instance?

ColinRedGriff, 2018-03-26:
I normally use a Wanhao i3 printer (Prusa i3 mk2 clone) which is when I noticed the issue. I added that printer when I installed the new version of Cura and copied the settings to it from my older version. The example above is a brand new printer profile I added yesterday to try and fix the problem, based on the custom printer option. 

I've tried several different flavours of gcode in that profile but they all seem to show the issue. 

I'm printing via a file not directly to the printer.

fieldOfView, 2018-03-26:
Can you load an object, and then save a project and post a link to that project here? That gathers up your configuration and we can try to reproduce the issue (and hopefully fix it, if it has not already been fixed recently).

ColinRedGriff, 2018-03-26:
Here is the link to the project
https://www.dropbox.com/sh/tdidp7tvr8fozzn/AAATjKi_N2So9s-h8WcNyEsua?dl=0

fieldOfView, 2018-03-26:
Hmm, I just loaded your project file into a 3.2.0 installation, and got the following gcode output:

```
;FLAVOR:Marlin
;TIME:2205
;Filament used: 1.4134m
;Layer height: 0.1
;Generated with Cura_SteamEngine 3.2.0
M82 ;absolute extrusion mode
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S245 ; set extruder temp
M104 S245 ; use material temp
M104 S250 ;hardcode 250 degress
M140 S60 ; set bed temp
M190 S60 ; wait for bed temp
M109 S245 ; wait for extruder temp
```

ColinRedGriff, 2018-03-26:
I'll try loading a copy of Cura 3.2.0

fieldOfView, 2018-03-26:
I don't expect much difference between 3.2.1 and 3.2.0, it is just that I still had 3.2.0 installed (I normally run Cura from source).

ColinRedGriff, 2018-03-26:
Seems to be OK in 3.2.0 on my Mac High Sierra 10.13.3.

Thanks

Ghostkeeper, 2018-04-20:
So ColinRedGriff's problem is resolved, but the original report from @Stickyfeatures still stands. I wasn't able to reproduce the bug so far...

Stickyfeatures, 2018-04-24:
Hi, I've been using a different printer so not able to retest but I'll return to using my Replicator soon. I'll give you a shout if the problem persists.

ianpaschal, 2018-05-29:
Haven't heard anything for more than a month so it seems to be fixed for everyone.

closed 2018-05-29 by ianpaschal

ducky92, 2018-10-07:
I am currently having this issue with cura 3.5.0. downloaded the installer from the website this morning. fresh install on win 8.1, loaded the prusa profile from https://manual.prusa3d.com/Guide/How+to+import+profiles+to+Cura/703. using the pet profile setting nozzle temp to 240C generates gcode that sets temp to 210C.

drgrandios, 2018-10-10:
This is a "me too" comment - I used Cura 2.7.0 before (via AppImage on Linux Mint). Today I switched to Cura 3.5.0 and the G-code always uses 225 degrees at startup. Only at the end of layer 0 the temperature switches to the configured setting (235 degrees in my case). 

The profile is based on Prusa i3 Mk2 with generic PETG selected as material.

drgrandios, 2018-10-10:
The info comment at the end of the generated G-code indicates there are 2 different material_print_temperatures:

```
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = Nunus\\ndefiniti
;SETTING_3 on = fdmprinter\\n\\n[metadata]\\nquality_type = normal\\ntype = qual
;SETTING_3 ity_changes\\nsetting_version = 5\\n\\n[values]\\nacceleration_enable
;SETTING_3 d = True\\nadhesion_type = skirt\\nbrim_width = 3\\ncool_fan_full_at_
;SETTING_3 height = 0.2\\ncool_fan_full_layer = 3\\ncool_fan_speed = 100\\ncool_
;SETTING_3 fan_speed_0 = 20\\ninfill_overlap = 30\\ninfill_pattern = lines\\ninf
;SETTING_3 ill_sparse_density = 20\\njerk_enabled = True\\nlayer_height = 0.24\\
;SETTING_3 nlayer_height_0 = 0.08\\nmaterial_bed_temperature = 80\\nmaterial_bed
;SETTING_3 _temperature_layer_0 = 80\\nmaterial_flow = 130\\nmaterial_print_temp
;SETTING_3 erature = 225\\nretraction_amount = 3\\nretraction_combing = infill\\
;SETTING_3 nretraction_speed = 70\\nskin_overlap = 30\\nspeed_print = 80\\nspeed
;SETTING_3 _travel = 200\\nspeed_travel_layer_0 = 70.0\\nspeed_wall = 50\\nsuppo
;SETTING_3 rt_angle = 75\\nsupport_enable = False\\nsupport_pattern = grid\\nsup
;SETTING_3 port_type = buildplate\\nsupport_z_distance = 0.12\\ntop_bottom_thick
;SETTING_3 ness = 1.2\\nwall_thickness = 1.2\\n\\n", "extruder_quality": ["[gene
;SETTING_3 ral]\\nversion = 4\\nname = Fine #2\\ndefinition = fdmprinter\\n\\n[m
;SETTING_3 etadata]\\nquality_type = normal\\ntype = quality_changes\\nsetting_v
;SETTING_3 ersion = 5\\nposition = 0\\n\\n[values]\\nbrim_width = 4\\ncool_fan_s
;SETTING_3 peed_0 = 30\\ninfill_overlap = 15\\ninfill_sparse_density = 10\\nmate
;SETTING_3 rial_flow = 110\\nmaterial_print_temperature = 235\\nmaterial_print_t
;SETTING_3 emperature_layer_0 = 234\\nskin_overlap = 10\\n\\n"]}
```

The first seems to be the one used in the startup code, the second is used later.

kamushadenes, 2018-10-10:
Can confirm on 3.5.0... very annoying when printing with anything other than PLA.

pr3sidentspence, 2018-10-12:
Cura 3.5. Printrbot Simple Pro. Created by adding a Simple and editing dimensions. Using necessary start/end gcode as per Simple Pro over USB instructions. Using default PLA Cura temperature setting. Prints start at 210 for 4 minutes then drop to 200 for remainder of job. Can see on Octoprint temperature graph that 210 then 200 are the requested temperatures.

KazaHesto, 2018-10-13:
Also getting the problem on Cura 3.5 on my Makerbot Replicator 2. Using the default generic PLA profile.

ducky92, 2018-10-19:
my issue seems to be resolved with the most recent update

drgrandios, 2018-10-21:
I agree with ducky92, the issue has been resolved (at least for me) when switching from Cura 3.5.0 to 3.5.1. Startup G-code contains the correct M104