Ultimaker/Cura #3183: 3.2 Beta won't slice STL model that isn't touching the build plate with "use adaptive layers"

BenMarolt, 2018-01-25:
Trying to print a sphere without getting a flat spot from touching the build plate. I was able to get good results in the past by positioning it above the build plate so that it prints entirely on support structure, but it will not generate anything printable, as in it makes gcode for only a single brief layer that only sets speed and acceleration and doesn't have any movement nor any extrusion.

Application Version
Ultimaker Cura 3.2 BETA
Downloaded from Ultimaker's website

Platform
Windows 7 Ultimate 64-bit

Display Drivers (2 video cards in this system)
NVIDIA GeForce GT 630 23.21.13.8813
NVIDIA GeForce GTX 960 23.21.13.8813

**Steps to Reproduce**
I have settings not to automatically drop to the build plate and not to automatically slice.
Load an STL model and change the Z to 1mm (or any arbitrary height above the build plate) then press prepare button to slice it. 

**Actual Results**
"Ready to Slice" text changes to "Slicing..." for 1 or 2 seconds, then goes to "Ready to Save to File", but the time estimate still says "00h 00m 0.00m / ~0g" and the layer view has nothing but a shadow of the model. There isn't an error displayed, but it's as if the engine is giving up immediately if there is nothing directly touching the plate. When this doesn't cause Cura to crash and close immediately, the generated gcode only contains start and end code with a single "layer 0" between that doesn't really have much to it, like this:
;LAYER_COUNT:0
G1 F900 E-4
M201 X4000 Y4000
M202 X4000 Y4000
M207 X20
M107
[2cmSphereShell_1mmZoffset.zip](https://github.com/Ultimaker/Cura/files/1662379/2cmSphereShell_1mmZoffset.zip)

**Expected results**
Support should be generated up to the model and there should be the regular layers above that.

**Additional Information**
I have adaptive layer height turned on, but if I untick "use adaptive layers" slicing works normally. With "Use adaptive layers" ticked, sometimes wildly out of range numbers appear for the min and max of the layer thickness. Sometimes Cura just crashes and closes out, sometimes it doesn't crash, but generates the "empty" gcode.
Here's my project and log:
[2cmSphereShell_1mmZoffset.curaproject.3mf.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1662382/2cmSphereShell_1mmZoffset.curaproject.3mf.curaproject.3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/1662384/cura.log.zip)

Also, I noticed a return of the issue with a repeated "curaprofile.3mf" extension that had been fixed a couple versions back. See also [Cura 2.4: Project's file extension #1485](https://github.com/Ultimaker/Cura/issues/1485)

BenMarolt, 2018-01-25:
@ChrisTerBeke Maybe you would have some insight on this considering your [pull #649](https://github.com/Ultimaker/CuraEngine/pull/649). Is this something that can be handled during calculation of the adaptive layer heights by setting it to use the regular layer height whenever there is no portion of the model within that same height from the current/prior Z during slicing?

ChrisTerBeke, 2018-01-25:
We know there's some edge cases with adaptive layer height that haven't been tested yet, this is apparently one of them. That's why it's in experimental for now. Thanks for sharing the project file. I'm not sure how high of a priority this has the coming months, but I'll at least create an internal ticket for it!

BenMarolt, 2018-01-25:
@ChrisTerBeke Thanks, I like playing with the experimental settings! I hope this info will be useful down the road.

Ghostkeeper, 2018-02-13:
Also happens with models that consist of multiple parts vertically. It'll slice the bottom part but not anything above the empty layer. It seems that the algorithm just stops at the first empty layer it encounters and forgets about everything that's higher.

Enverex, 2019-05-03:
I've just encountered this issue, so it's still present with Cura 4.0.

My example use-case: I needed to print the model entirely on supports, hence raising it above the bed.

diegopradogesto, 2019-09-05:
We were taking a look at this issue and after some trials and errors, we've decided to defer this for now at least.

falo2k, 2020-06-04:
Still very present in Cura 4.6.1 as just ran into this.  Aiming to fully support a model, but can't do this with adaptive layers.

spiralslice, 2021-04-28:
This problem is still present in 4.9: I have an object that needs curved support on the underside, but when I raise it off the buildplate by 3mm, the Slice button gives me an empty gcode file.

I got around it by putting another small object directly on the build plate, and then Cura slices everything properly, including my floater. I will probably cancel the dummy object through OctoPrint's "Cancel Objects" plugin. 

It's not the worst workaround, just a silly thing to remember.

andygomez3dart, 2021-09-16:
Still present in CURA 4.10.0

closed 2023-06-24 by github-actions[bot]