Ultimaker/Cura #3184: Suggestion: Please consider double clicking center mouse button for reset/center view

Thisismydigitalself, 2018-01-25:
Like with F360, clicking center mouse button twice will reset, center and fit model to screen.

Ghostkeeper, 2018-02-13:
There's a button at the top of the screen for this. Is that not easy enough?

Thisismydigitalself, 2018-02-13:
It does not fit the model to screen, so - one needs to move mouse cursor to button, click on button, than click again the part cause even thou it looks like it is selected (zoom will not center it to screen), so select the model again than zoom in to fit & center.

With Fusion360 - double click center mouse, Voila!

Ghostkeeper, 2018-03-08:
Our project manager put this off of our planning. We won't be getting around to this any time soon.

Thisismydigitalself, 2018-03-08:
Thanks for this update. 
Just wondering -  if there is a button already for this, why should this take long to just change the parameters for the built plate to fit the screen better?

Ghostkeeper, 2018-03-26:
> It does not fit the model to screen, so -

In fact the button moves the camera to a hard-coded position, regardless of how big your printer or your model is.

AdrianoFerrari, 2018-07-21:
In almost all cases I've seen, it'd be much easier to find and center the model if the camera's rotation origin were at the center of the **build plate**, instead of the center of the printable cube/volume.

Just poking around, but it seems that here is the line that would need to be changed:
https://github.com/Ultimaker/Cura/blob/9856ccf70030d365fc02dd9f08c436997edaed16/cura/CuraApplication.py#L784

```diff
-        camera_tool.setOrigin(Vector(0, 100, 0))
+        camera_tool.setOrigin(Vector(0, 0, 0))
```

Thoughts? I'll open a separate issue for this ("Set camera origin to middle of build plate"), if you'd like.

Ghostkeeper, 2018-08-31:
> In almost all cases I've seen, it'd be much easier to find and center the model if the camera's rotation origin were at the center of the build plate, instead of the center of the printable cube/volume.

For small models, yes. For large models, no.

> PS: I just found the "Center camera when item selected" option under "Preferences > Configure Cura > General". A simple UX fix would be to make this the default, as it makes it much easier to navigate and view the model.

I know! I enable this preference always. It's much more handy. Only our support department got feedback that it's annoying that the camera keeps moving automatically every time you click on an object, so we decided to disable it by default.

By the way, a feature has been implemented for the next version that makes the default camera position dependent on the size of the build volume, so it should default to a zoom level that allows you to see your entire printer.

AdrianoFerrari, 2018-08-31:
>  It's much more handy. Only our support department got feedback that it's annoying that the camera keeps moving automatically every time you click on an object, so we decided to disable it by default.

Alright, understood. Thanks for getting back to me on the reasoning behind these choices and defaults. :slightly_smiling_face: