Ultimaker/Cura #3185: Crashing (at Application Launch) on Windows 10 Pro - ALL VERSIONS

barrettdent, 2018-01-25:
**Application Version**

15.04.6 / 2.x / 3.x

**Platform**

Windows 10 Pro / Version 1709 / OS Build 16299.192

**Qt**

All from Ultimaker Site

**PyQt**

**Display Driver**

Intel Corporation
7/24/2017
Version 22.20.16.4749

**Steps to Reproduce**

Install, Run (CRASH)

**Actual Results**

15.04.6 - "pythonw.exe has stopped working" (dialog, self-exit)
2.x / 3.x - crash and self-exit after splash screen without any message

**Expected results**

**Additional Information**

I've tried compatibility modes. Tried giving admin privileges to Cura, python, pythonw, etc.

Running on a Intel Core i7-7700HQ CPU @ 2.80GHz / 32.0 GB RAM / 64-Bit OS

The exact same version of 15.04.6 runs fine on my older Windows 7 laptop, I've seen numerous complaints from Windows 10 users that go unanswered or get "it works on my machine" responses. I expect more. This computer runs SolidWorks 2017, Inventor Pro 2017, AutoCAD 2017 all with no issues. So why on earth can't I run any versions of a pathetic little slicing engine?  I see new features coming out. But I don't see anyone resolving this on-going nightmare that seems to affect a percentage of users.

ChrisTerBeke, 2018-01-25:
When I Google for `pythonw.exe has stopped working` it seems it appears with a lot of Python applications, but no real solutions appear. I don't think it's related to Cura though, but rather a system error in general.

barrettdent, 2018-01-25:
If you read carefully the pythonw.exe relates ONLY to 15.04.6.  The newer versions just crash inexplicably.

Yes there is other junk software out there with pythonw.exe issues. Notably some TV app (which I do NOT have.) But there's also plenty of software that works just fine on Windows 10 Pro too.

Should I be unable to slice anything because nobody wants to even try to figure this out? Is it okay that none of the versions of your software works on a Windows OS?  No and no.

I've gone out of my way to give you as much info as possible. I'm open to trying to log stuff or whatever. Tell me what you need and how to get it. Because it's ridiculous not getting this resolved isn't an option. I've waited months. I've put in effort to try to find a resolution. But I'm not a programmer. I've gone as far as I can.  But I'll help if I can.

It's just not okay that NONE of the versions of your software seem to be able to run without crashing. I'm not alone. I'm tired of seeing folks with this problem brushed off.

fieldOfView, 2018-01-26:
> Should I be unable to slice anything because nobody wants to even try to figure this out?

No, not at all. But being obnoxious here does not help your case.

FusionBela, 2018-01-26:
I have the same problem, i opening a model and next Cura 3.1.0 crashed. (CuraEngine.exe)

jackha, 2018-01-29:
I also had Cura 2.x / 3.x crashing after the splash screen. While I used 2.x before it didn't work anymore. It seemed that my windows needed an update badly and after the update all was fine again. Hope this helps.

biscuitlad, 2018-01-31:
Not sure it's related, but it sure sounds like I had the same crash, but the splash screen doesn't even load. This is on Ubuntu 16.04, Cura 3.1. i7 platform. 2.7 works ok.

alekseisasin, 2018-02-01:
@biscuitlad  After upgrading Cura version, it tries to trigger version upgrade functions and might be for some mystery cases it fails and certain data is corrupted.

jackha, 2018-02-05:
In the upcoming 3.2 version, when an upgrade results in a crashing Cura, then it asks if the user wants to clear the preferences and start clean. Old settings are backed up.

Ghostkeeper, 2018-02-27:
`pythonw.exe` crashing happens when the Python interpreter crashes, which could happen either due to bugs in CPython or (more likely) due to an imported external binary library which crashes. Since Legacy Cura would just run the CPython interpreter separately on .py files it was `pythonw.exe` that crashed. Nowadays it would say `Cura.exe` since we pack CPython and the .py files inside the Cura.exe binary.

> I'm open to trying to log stuff or whatever.

That would help! You can find the logs over at `C:\Users\<you>\AppData\Roaming\cura\3.2\cura.log`. Perhaps you could post the last 50 lines or so of that file here?

JimskyKorsokov, 2018-04-09:
Hello, I'm having problems too that sound familiar to what is given above. I installed CURA 15.04.3 since that's what came with my printer. On opening it first time after install it asks if I want to upgrade to which I respond No. I then use the software perfectly happily but once it is shut down I cannot re=open it - says not responding - on force quitting it tells me there's a problem with PythonW.EXE. I tried deleting it and installing the latest version but that will not launch either. After uninstalling everything and re-installing from my printers disk it will run again - but not relaunch after shut-down.

I'm on Windows 10 and it seems something is now corrupted along the way. I do see Pythonw attempt to start under Processes when launching Cura, but it disappears as soon as the update window opens.

It's a great programme - just a pain to de-install / fresh install every time I want to use it.

Regards,

Jim

ChrisTerBeke, 2018-04-09:
As indicated above several times, when this happens, please share the Cura logs. Without those we can't begin to understand what is going wrong in your case.

JimskyKorsokov, 2018-04-09:
Hi, and thanks for the really quick response. I can find a log file for Version 3.2 but not for 15.04.3. Any clues as to where it might be?

 

Apologies,

 

Jim

fieldOfView, 2018-04-09:
15.04 is not supported here

JimskyKorsokov, 2018-04-09:
Hi - I have uninstalled 15.04.2, restarted my computer and installed 3.3. It fails on launch - see screen captures - I also attach the log file as requested. All help very gratefully received - I fear I've got some legacy stuff that needs deleting.
 
[image]
[image]
[image]

[cura.log](https://github.com/Ultimaker/Cura/files/1891728/cura.log)

JimskyKorsokov, 2018-04-17:
SOLVED .... but it's a weird one. (Version 3.2.1 and 3.3 BETA)

I have 2 monitors on my Windows 10 computer. When I launch CURA I drag the window over on to my bigger monitor - next time I touch the interface I get an instant crash. If I do not drag the window over - it works.

I've now unplugged one of my monitors and everything works fine. Hope this is of help to somebody somewhere.

Jim

Ghostkeeper, 2018-05-04:
We've sometimes heard that this could be caused by having two graphical cards, each assigned to one monitor. It was never confirmed though. It could also be caused by having two different pixel scaling factors in your monitors (like, one of them is HDPI and the other isn't). But we don't know because we haven't been able to reproduce that crash.

fieldOfView, 2018-09-26:
Could you try keeping Cura on the primary display?

dgompert, 2019-11-09:
I have an HP 6570 laptop that runs ever other application and ran Cura for a couple of days after I rolled back the Intel video driver.  Now Cura hangs and won't even start.  It also is hung across multiple virtual desktops.  
The system has all standard equipment except an SSD and 12 GB of ram.

closed 2020-07-03 by no-response[bot]