Ultimaker/Cura #3186: FeedRate Limits - Suggestion.

Thisismydigitalself, 2018-01-25:
I really like this FeedRate color map.
Typo or slicing algorithm can lead to a higher speed one's printer/filament can handle. ask me how I know. It would be nice if this very nice feedrate color spectrum map be more like an audio VU meter where 0 DB is top level and higher than that will be marked with +/red/clipping/checkerboard/etc/ so it will be obvious that somewhere along the print top speed was exceeded. these days you can get to higher than 100mms which will stretch the color map accordingly. this will not give us a clear indication about a possible issues along the print.

Just for reference - should follow Cura's current color scheme.
[image]

A penny for your thoughts.

jackha, 2018-01-29:
Thanks for your idea, nice VU meters ^^. However, we don't really know the actual top speed while 0 dB is a hard limit. Also high speeds are mostly not actually reached and that's influenced by acceleration.

Thisismydigitalself, 2018-02-12:
"Speed" was a bad choice of word on my behalf  - I meant Frequency.  I wish for this meter to show me when and where my nozzle will vibrate most violently, e.g. where speed is confined to very short distances.

Ghostkeeper, 2018-03-07:
Detecting vibrations occurring is quite hard though. It basically requires us to simulate the hardware. Doesn't it? The vibrations will be vastly different for every printer.

Thisismydigitalself, 2018-03-07:
True and yet a unit-less color map showing the longest to shortest line's travel weighted time (red would be the shortest / HiFreq) should give us an idea of what will be going on. more often than not i see my printer's head, during action, moving violently while printing a small hole or trying to build a bridge support, etc.

Ghostkeeper, 2018-03-30:
We could visualise the expected acceleration magnitude in a vertex. That would come close, right?