Ultimaker/Cura #3188: Slicing gets stuck half way through

ChrisRiddell, 2018-01-26:
**Application Version**
3.2 Beta

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Load one of these STL files from https://www.thingiverse.com/thing:2449890 then let it slice

**Actual Results**
Slicing bar stops at 50% through and never goes any more

**Expected results**
To fully slice

**Additional Information**
[cura.log](https://github.com/Ultimaker/Cura/files/1666375/cura.log)

smartavionics, 2018-01-26:
I can slice either of those OK. Even with the wall order optimization enabled, the longest it took to slice was around 30 seconds. What profile are you using?

ChrisRiddell, 2018-01-26:
Stock UM2+ with PLA or ABS materiel selection and any quality profile.

Thisismydigitalself, 2018-01-26:
sounds similar to my experience.
https://github.com/Ultimaker/Cura/issues/3168

ChrisRiddell, 2018-01-26:
After clearing the appdata folder it was faster but still took awhile.

[cura.log](https://github.com/Ultimaker/Cura/files/1667092/cura.log)

So far it's only been this model.

ChrisRiddell, 2018-01-26:
[UM2_B_PolySher_Circular_Holed_Platform_.curaproject.zip](https://github.com/Ultimaker/Cura/files/1667142/UM2_B_PolySher_Circular_Holed_Platform_.curaproject.zip)

Here is a project file if that's any help.

smartavionics, 2018-01-26:
OK, I've found where the slowness is coming from - it's related to the optimization of the skin lines. I will work on a fix for the next beta. Sorry for the inconvenience.

ChrisRiddell, 2018-01-26:
Is that the optimization walls? as that settings is off.

smartavionics, 2018-01-26:
No, it's different to that.

smartavionics, 2018-01-26:
I had https://github.com/Ultimaker/CuraEngine/pull/693 already as I guessed there could be a speed problem but I thought I would wait until a real life case occurred. Hopefully, they will put the fix into the next beta or release.

closed 2018-02-05 by ChrisRiddell