Ultimaker/Cura #3189: SideBar slow when all settings visible

ChrisRiddell, 2018-01-26:
**Application Version**
3.2 Beta

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**
AMD 17.12.2

**Steps to Reproduce**
Unhide all the settings in the custom section of the sidebar 

**Actual Results**
Very laggy when opening some categories for eg cooling opens smooth but speed does not.

**Expected results**
for all categories to opening smooth.

**Additional Information**
https://youtu.be/pw1at52ty3Y

Ghostkeeper, 2018-02-13:
Yes, this reproduces for me too. The categories that are slower tend to have more settings in them (hidden or not) so that is an indication.

We'll be upgrading to Qt 5.10 with our next version, which has functionality to make our theming much faster to render. We need to transform all the QML for this though so we'll be refactoring bits and pieces one by one to the new structure.

Ghostkeeper, 2018-02-13:
I think I'll close this issue in favour of #3221. Your ticket was first, but the other has more information in it so far.

closed 2018-02-13 by Ghostkeeper