Ultimaker/Cura #3190: GcodeViewer doesn't display first line after extruder switch

MB3Y, 2018-01-26:
**Application Version**
3.1

**Platform**
Win10

**Steps to Reproduce**
File, Open, select Some.gcode containing T0, T1, ... commands

**Actual Results**
GCode is displayed, but first line after an extruder switch is not shown.

**Expected results**
All gcode lines should be depicted.

**Additional Information**
In the picture an edited gcode file - each 36 degrees an extruder command is inserted ( 8 times T0,  times T1 I believe) You notice tiny empty spaces - that is the gcode line not visualized.
[image]
For short lines this is maybe not such a problem, but for longer lines it disturbs the view pretty much.
[image]

Vandrasc, 2018-01-26:
Hi,
I am trying to reproduce your issue with the given scenario but my layers are being correctly generated. What printer did you use, and did you changed the gcode flavor?

MB3Y, 2018-01-26:
As I indicate above the first picture: I do slice with Cura first, then edit the GCode manually.
I.e. I add T0, T1, T2 commands in between the existing GCode lines. No more, no less.
But your comment put me to some deeper analysis and apparently this does occur when the Tx command is inserted between two G1 commands. When the G1 command is only laying a very short track it will not be (so) visible, but if you insert extruder change before a longer one, then it becomes very visible.

Vandrasc, 2018-01-29:
@Ghostkeeper, would you mind have a look, is this expected? Thank you :)

Ghostkeeper, 2018-02-14:
No, I wouldn't expect this.

Maybe the current E value is messed up when switching to the other extruder.

Ghostkeeper, 2018-02-14:
Actually, we don't know where the nozzle ends up after the extruder switch because its position might change in the process of switching. This could be complicated. In a normal g-code file (not modified by the user) the nozzle would be moved away from the print (sometimes to a nozzle switching bay or so) and then move back. These moves seem to be displayed by layer view for me at least, but the move from before the switch to after the switch isn't. It could be on purpose?

MB3Y, 2018-02-14:
FYI: discovery of the evening - but mystery not yet resolved (I'm still searching: why does all work when cura generates multi-extrusion model??):
So: adding a Zhop a after each extruder switch improves much in the visualisation, but shows smaller line thickness. Only sometimes the line is completely missing.

So my insert in a default gcode file is
"T<extrudernr>"
"G0 Z<current layer height+0.01>"
"G0 Z<current layer height+0.01>"

[image]
[image]

MB3Y, 2018-02-25:
@Ghostkeeper 
> Maybe the current E value is messed up when switching to the other extruder.
if it really is about extrusion messed up, then I would expect the situation be resolved changing the code as below - but also then it doesn't show the fist line afte the second G92 cmd.

orig code
  G1 X<xxx1> Y<yyy1> E<extr1>
  G1 X<xxx2> Y<yyy2> E<extr2>

new code
  G1 X<xxx1> Y<yyy1> E<extr1>
  G92 E0
  T<tt>
  G92 E<extr2>
  G1 X<xxx2> Y<yyy2> E<extr2>

Ghostkeeper, 2018-03-14:
We do put a G92 command in there at every extruder switch, I believe.

    G92 E0
    T1
    G92 E0
    G1 X41.221 Y102.510 E0.01126

diegopradogesto, 2019-09-10:
Are you still experiencing this problem. We couldn't reproduce it internally, maybe because it's printer dependant (because of some settings or gcode flavor). 

If you still want us to investigate, please provide a project file with this specific scenario that we can use to generate the gcode (or send a GCode file)

MB3Y, 2019-09-16:
@diegopradogesto : I haven't tested this in the Cura 4.x gcodeviewer versions yet, but your argument that it could be printer dependant or gcode flavor dependant is incorrect simply because of how it's reproduced: generate gcode with Cura - edit manually by simply adding Tx cmds in between (nothing more nothing less) - import in Cura again.
There is nothing printer specific there - or it has been generated by Cura in the first place?

Ghostkeeper, 2019-09-17:
The g-code displayed when importing a g-code file makes the assumption that it's g-code that would be printed by the printer that you currently have active. This is necessary for a couple of missing bits of information that are not in the g-code. From the top of my head:
* Filament diameter
* G-code flavour
* Nozzle offset

There's probably more of these small things it needs to know for the printer to reconstruct the model from the g-code.

closed 2020-06-19 by no-response[bot]