Ultimaker/Cura #3191: Model placement wrong when opening a project file

daboi, 2018-01-26:
**Application Version**
3.2.0 BETA

**Platform**
macOS 10.12.6 (16G1114) (Darwin 16.7.0)

**Qt**
Builtin one 

**PyQt**
Builtin one 

**Display Driver**
NVIDIA GeForce GT 750M 4.0.8 [3.2.8]

**Steps to Reproduce**
I created a build plate placement with my model then saved as project file. When I opened it again model places were wrong. I fitted 8 models in my plate but when I open only 4 of them are on plate. 

**Actual Results**
Can only see 4 models on plate and some off plate. 

**Expected results**
All 8 models should be placed in their location on bed. 

**Additional Information**
I grouped models into 2, maybe that might do difference. 

[image]

Project file
[CCR10_BagClip_85.3mf.zip](https://github.com/Ultimaker/Cura/files/1669179/CCR10_BagClip_85.3mf.zip)

Thisismydigitalself, 2018-01-27:
I tried loading your project. it might look similar to your findings but if you look closer you'll see the project did not open the same on my two versions - Settings are not the same with 3.1 as with 3.2.

With 3.1R
[image]

Winth3.2B
[image]

daboi, 2018-01-27:
Then it might be a problem on saving the project?

Emre

[31]<https://user-images.githubusercontent.com/30949968/35470901-4028797c-035a-11e8-9491-de68edc4dd3e.png>

[32]<https://user-images.githubusercontent.com/30949968/35470903-448cd350-035a-11e8-91c1-ee455b1d1c30.png>

Thisismydigitalself, 2018-01-27:
Dono. As we were told, the guys are rewriting and untangling some wires to improve things.
When it comes to Project behavior i sure like to hear what the UX and product designers have to say about it cause maybe it's us that don't use it the way it was designed to work.

Thisismydigitalself, 2018-01-27:
Wait, this is not a project. did you save/export it as a project?

daboi, 2018-01-27:
Yes, I did. I used Save Project option from File menu.

Thisismydigitalself, 2018-01-28:
I believe Projects includes  ***.curaproject**.3mf suffix added to the file name.
I just tried arranging some stl's in Cura and save it as a project. when I reopened my project the stl's where place in a different location on the print bed. 

Not good.

Ghostkeeper, 2018-02-21:
On MacOS, the extension is only .3mf since MacOS's file dialogue doesn't handle the double extension well.

The 3MF file at the initial post of this ticket though is just 3MF, not a Cura project file. You can see that by opening it up with a Zip archiver. If it's a Cura project there should be a folder "Cura" in the root of the zip archive which contains the profiles and preferences that were used. Cura should be able to read both files with .curaproject.3mf and with .3mf as extension as project files. It just checks for that folder.

We've tried with MacOS, but "Save Project..." properly saves a project file. "Save As..." and selecting 3MF as file type will save it as a 3MF file.

daboi, 2018-02-21:
@Ghostkeeper With current Cura 3.2.1 I tried to load same files and clicked Save Project with 3mf extension. And unzipped and checked Cura folder exist. Loaded as project and objects got placed correctly. So this problem seems to be solved.

Thisismydigitalself, 2018-02-21:
Please follow this my post of mine and check again. i will do the same. https://github.com/Ultimaker/Cura/issues/3201#issuecomment-364128938

diegopradogesto, 2018-07-03:
Hey @daboi , is the original issue fixed? In this case we can close the ticket.

daboi, 2018-07-03:
yes

closed 2018-07-03 by diegopradogesto