Ultimaker/Cura #3192: Feature Idea: Allow profile selections in Per Model Settings

filipgoc, 2018-01-27:
It would be great if we could simply select a different printing profile (preset) per model instead of having to select all the parameters we want to modify and in what ways. 

I'm not sure if I need to argue more to provide use cases etc. It seems pretty self-evidently useful, but maybe it's just me. 

Happy to elaborate if this gets no ❤️

jackha, 2018-01-29:
Sounds nice, but of course not all profile settings can always be applied. Most importantly, the layer thickness.

Ghostkeeper, 2018-02-14:
> Cura could filter the profiles to only show those with the same "quality_type". That would alleviate the layer height issue, but you would still have things like bed temperatures, or bed adhesion, which cannot be set per model (or only when printing one at a time), so you would not get the full profile (and thus unexpected results).

Apart from the issues you described, the user can also change the layer height in a custom profile without the `quality_type` changing. Normally this custom profile is saved with one container for each extruder and global and applied to each stack at the same time, so the layer height is then correct even if the layer height is different from what it would normally be for that `quality_type`. But if the `quality_type` is unchanged but the layer height was different, then you could still have different layer heights per model which then fails to slice as such.

> But these are the same. The filter for allowing 'presets' is the same filter applied to settings to make the 'per-model-allowed-settings' subset.

Not really. The filter for allowing to change settings per model is to check the "settable_per_model" metadata item. The filter for allowing to select custom profiles per model would be something like "all of the settings that get overwritten by this profile are settable per model", which will need a new routine.

> Use case one:
> I have multiple meshes. I go through the painstaking process of modifying pre-model-settings of some. Then, I acquire a new version of one of these meshes. I want to simply replace the mesh and keep the per-model settings. As of now, I'd have to re-create all the settings from scratch....

If you refresh (F5) then I'd expect the per-object settings to remain in place. I just tested that and it seems to work correctly.

filipgoc, 2018-02-14:
> If you refresh (F5) then I'd expect the per-object settings to remain in place. I just tested that and it seems to work correctly.

@Ghostkeeper wait, is there a 'replace mesh' somewhere? That would be amazing. 
I don't see it in the menu nor in the rightlick menu. 

For me, replacing simply meant
- deleting the old mesh,
- and importing a new one that would serve the same function (but update, so different).
- As such, Cura correctly treats the new mesh as... new,
- and so refreshing does not bring back anything.

Ghostkeeper, 2018-03-07:
F5 causes the mesh to reload from file, replacing the model in Cura in place. If you had a mesh on your build plate, positioned in just the right way, scaled, rotated, per-object settings, whatever, it would retain all of that but use the new file as source (so the new file also gets rotated and such). It's under the context menu called "Reload All Models".

It doesn't work for project files though and also not for images.