Ultimaker/Cura #3194: Python install paths violates Fedora and (open)SUSE packaging guidelines

StefanBruens, 2018-01-27:
**Application Version**
Cura 3.1.0

**Platform**
Fedora
openSUSE

**Steps to Reproduce**
build cura or uranium

**Actual Results**
Pure python modules are installed in 
` /usr/lib64/python3.6/site-packages/cura/ `

**Expected results**
` /usr/lib64/python3.6/site-packages/cura/ `

**Additional Information**
On at least Fedora and openSUSE, installation directory depends on the architecture *and* distinguishes between pure python (noarch) and arch dependent modules.
 
https://fedoraproject.org/wiki/Packaging:Python
https://en.opensuse.org/openSUSE:Packaging_Python#File_lists
https://docs.python.org/3/distutils/apiref.html#distutils.sysconfig.get_python_lib

Probably best method would be to allow specifying the directories using a PYTHON_PLATFORM_INSTALL_PATH and PYTHON_INSTALL_PATH variable, with a fallback to the current implementation.

StefanBruens, 2018-01-27:
This got broken in https://github.com/Ultimaker/Cura/commit/aa92d4d0f97bcd36ccd1a7cfff30fd5fc68d843e#diff-af3b638bc2a3e6c650974192a53c7291

Ghostkeeper, 2018-02-13:
For some things that we install, this cannot be detected. We don't know whether a plug-in is pure Python or has binary stuff in it.

For the stuff that we do know, we could make an exception for OpenSUSE and Fedora to not use `${LIB_SUFFIX}` for those things?

nallath, 2018-02-14:
I'm running it on fedora, so if there are any change, I can test them.

StefanBruens, 2018-02-18:
All off uranium and cura (not cura-engine) is pure python, which is *known*, no reason to "detect" this. This includes the python modules (i.e. stuff installed to <...>/site-packages/) and plugins.

For (3rdparty) plugins this is of course not known, but this is cared for by https://github.com/Ultimaker/Cura/commit/6eeaab3c12fe0d60f8e841927cb7b209b482d916 , which looks for plugins in all plausible locations.

Btw., as far as I can see, Mageia as well install noarch packages (e.g. pure python) into /usr/lib/. I could not find anything about Arch (tried to come up with a good search term - unfortunately "arch" is somewhat unspecific ...).

I still think using a  CMake variable would be best, this would avoid doing any "magic" inside CMakeLists.txt. Its trivial for packagers to pass in the right path.

closed 2018-03-02 by ChrisTerBeke

StefanBruens, 2018-03-02:
Whats the reasoning behind this? Do you prefer every distribution adds their own patches?

ChrisTerBeke, 2018-03-02:
Mostly because our Linux user base is relatively small (less than 2%), and Fedora even more so. We'd rather be clear on that we won't put effort into it. A pull request is always welcome of course.

StefanBruens, 2018-03-02:
How do you measure your user base?
This is not only about Fedora, but also openSUSE, Mageia, RHEL, SLE, ...
And "Won't Fix" is the most stupid and repressive bugfix "solution" I am aware of - of you want contributions, don't do that!

ChrisTerBeke, 2018-03-02:
We measure quite accurately from the user statistics. Won't fix is just a label to indicate that we (as in Ultimaker) won't fix it. Keeping 700 issues open is not feasible, so we're just doing some cleanup. Also, Cura is Free and Open Source Software, so if you're not contributing, just accept it as it is.

StefanBruens, 2018-03-02:
And you confirm the accuracy of the statistics by what?

About contribution, do you think a detailed bug report just manifests by itself? I have invested time for figuring out which distributions are affected, which commit broke it, giving references to relevant python documentation.
I wont do it the next time, as the time is clearly wasted, and just patch any issues downstream. If asked, I could have added a PR fixing the issue, but given your mindless behaviour, I will invest my time into more cooperative projects.

Ghostkeeper, 2018-03-28:
> All off uranium and cura (not cura-engine) is pure python, which is known, no reason to "detect" this. This includes the python modules (i.e. stuff installed to <...>/site-packages/) and plugins.

Good point. And we can't really install 3rd party plug-ins installed via the plug-in browser to /usr/lib because this is happening at runtime and would require administrator rights anyway.

I'd like to make a change for Fedora and OpenSUSE in Cura and Uranium. That would take 2 minutes. But I'd need a way to detect that the Linux distribution is Fedora or OpenSUSE. Does calling `/etc/fedora-release` work or something?