Ultimaker/Cura #3197: Cura 3.1 - Fill Gaps Between Walls extrudes into an existing wall line

Thorinair, 2018-01-27:
**Application Version**
3.1

**Platform**
Windows

**Steps to Reproduce**
1. Load the following model: https://www.thingiverse.com/thing:2705552
2. Make sure that you have a 0.4 nozzle and that Fill Gaps Between Walls is set to everywhere.

**Actual Results**
When filling gaps, Cura will cause extrusion directly into the middle, already existing, printed wall line. This will cause the extruder to clog up as it has no open space to extrude into. This happens on the vertical walls of the model linked above. Check image below to see which line gets filled incorrectly.

[image]

**Expected results**
Cura should not be filling an already existing wall. The width of these walls on the model is exactly 2 mm. This should be normally filled with 5 wall lines with no gaps needing to be filled. Maybe a rounding issue?

smartavionics, 2018-01-27:
It doesn't look to me that it is filling the gap because that is normally shown in yellow. More likely what is happening is that the middle wall (innermost green wall) is not being sufficiently narrowed. The way the overlap compensation works is that when a wall can't fit in the available space it is narrowed. In this model because the part is 5 wall lines thick, one of the wall lines needs to be thinned down to nothing (remember the walls always come in pairs and you are using 3 walls so that means it will need to print 6 lines, one of which would need to be zero width). It's failed for some reason, the overlap compensation is pretty buggy so that doesn't surprise me. Sorry, I don't have a solution to the problem right now.

Thorinair, 2018-01-27:
It actually is filling it. I know because I printed and observed my printer closely as it started clogging up. Sad that there is no solution...

smartavionics, 2018-01-27:
I use a 0.4mm nozzle and I nearly always use a line width of 0.5mm. I believe that it is normal to use a line width that is > the nozzle width. Anyway, if you used a line width of 0.5mm you should get what you want.

smartavionics, 2018-01-27:
Just tried that, it slices fine with a line width of 0.5mm with either 2 or 3 walls.

Thorinair, 2018-01-27:
Theoretically, this issue would happen again on a 2.5mm thick wall, when using a 0.5mm line width, no?

smartavionics, 2018-01-28:
That's right. It has to print a wall using an even number of lines. The overlap compensation is meant to adjust the width of the last line so that it will fit in the gap in the middle but, unfortunately, as I mention above, the overlap compensation has some problems (as discussed in https://github.com/Ultimaker/CuraEngine/pull/586). So, if possible try and use a line width such that an even number of lines will fit into the wall.

smartavionics, 2018-01-28:
One other thing, check in your profile that Compensate Inner Wall Overlaps hasn't been turned off. If it has that would not help at all. Even though the compensation is buggy, no compensation really would be bad when trying to print 6 lines where there is only space for 5!

Thorinair, 2018-01-28:
Yep, Compensate Wall Overlaps makes exactly 0 difference in this case. This definitely looks like a bug to me that should be fixed. One shouldn't be forced to use a weird line width to compensate for strange behaviour in Cura. Cura should be able to recognize that there are 2 walls over each other and simply just not print one of them.

smartavionics, 2018-01-28:
Rather embarrassingly, I think the problem you are seeing here is actually in my wall optimization code. There are bugs in the overlap compensation but this problem appears to be related to the wall optimization. I am currently trying to find where it's going wrong and will report back when I know more. Sorry for the inconvenience.

Thorinair, 2018-01-28:
Thank you for looking into it! :)

Thorinair, 2018-02-08:
The issue still persists in Cura 3.2, meh...

smartavionics, 2018-02-08:
Yes, my PR has not yet been vetted/accepted by the Cura devs. Hopefully it will make it into 3.3.

smartavionics, 2018-02-08:
As a workaround, I recommend using 2 walls rather than 3 if that is acceptable.

smartavionics, 2018-03-27:
Still waiting for https://github.com/Ultimaker/CuraEngine/pull/695  to get reviewed and merged.

smartavionics, 2018-04-25:
Hi @jackha , I am not getting the same results so I am probably not using the right settings, please post a 3mf.

jackha, 2018-04-25:
@smartavionics Thanks for looking at it :-) I just added an extra ".zip" in the file I used:

[CFDMP_SSD_Mount_v2.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1947548/CFDMP_SSD_Mount_v2.curaproject.3mf.zip)

jackha, 2018-04-26:
Ah! Thanks for your detailed explanation ^^.
We'll have a look at the new PRs soon and close this issue as is.

closed 2018-04-26 by jackha

jackha, 2018-04-26:
I'll also make an issue for the slice delay. On my computer it's a little less than 5-6 seconds, but it's still annoying.