Ultimaker/Cura #3198: Cura 3.2b significant 3D performance drop compared to 3.1

Spirou42, 2018-01-27:
**Application Version**
Cura 3.2b

**Platform**
MacOS 10.13.3 (17D47)

**Display Driver**
MacOS internal AMD FirePro D700 6 GB

**Steps to Reproduce**
load any model into Cura 3.2b
rotate or zoom the model

**Actual Results**
significant drop in frame rate.

**Expected results**
fluffy animation 

**Additional Information**
Have a look at https://youtu.be/05LhF0BR3PY 
Model used in video: [BasePlate-Steel.3mf.zip](https://github.com/Ultimaker/Cura/files/1670565/BasePlate-Steel.3mf.zip)

ChrisRiddell, 2018-01-29:
I'm also seeing this

Platform
Windows 10

Display Driver
AMD RX 580, Radeon 18.1.1

Ellecross, 2018-01-29:
I cannot reproduce this on my Windows 10 + Intel graphics. hmmm. 

@Spirou42 I did notice however, in your project file, the profile is not supported. Was this intentional? 

[image]

jackha, 2018-01-30:
@Ellecross I think it's the "material_type" being changed into something that's not supported by default.

Spirou42, 2018-01-30:
@Ellecross, @jackha It was. I had initially created this project for #3088 and recorded the video at the same time. Therefore i also attached the "faulty" project to the current issue for completeness as it was the project used in the recording.

jackha, 2018-01-31:
We have it on or radar to look at the laggyness. We actually found somebody at the office who has a laggy Cura. Thanks for sharing the info.

Spirou42, 2018-02-01:
@jackha Hey, lucky you. I know from first hand experience how difficult it can be to reproduce user issues.

Spirou42, 2018-02-21:
Checked in 3.2.1, works fine for me. Closing

closed 2018-02-21 by Spirou42