Ultimaker/Cura #3199: can't use removable drive for ultimaker 2, errno 13, no permission

Riagel, 2018-01-28:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Vandrasc, 2018-01-29:
Hi @Riagel ,

I do not think this is related to Cura, it seems you have insufficient permission, do you have all the rights and running as administrator?

fieldOfView, 2018-01-29:
> running as administrator?

Noooo! Don't run Cura as administrator!

@Riagel, we need *much* more information to be able to help you. Please fill in the template.

Vandrasc, 2018-01-29:
Hey, that was the next step ^-^, thank you @fieldOfView

ianpaschal, 2018-02-27:
No information was provided. Going to consider this forgotten.

closed 2018-02-27 by ianpaschal