Ultimaker/Cura #3201: Cura's "Save Project"

Thisismydigitalself, 2018-01-28:
With all due respect, Can Cura's UX / Product manager tell us all what "Save Project" was designed to save?Thank you.

jackha, 2018-01-29:
In short:
- The 3d model(s)
- Cura's printer, print profiles and settings

When loading a project file, you should get the complete state.

smartavionics, 2018-01-29:
Bu what isn't saved is the paths to the model files so that having loaded a project, hitting F5 doesn't reload the models (see https://github.com/Ultimaker/Cura/issues/2124).

Thisismydigitalself, 2018-01-29:
Thanks,
FYI: With both 3.1 and 3.2, saving a project with stl models arranged in a particular manner on bed, when reopening the project they are placed in a completely different layout.

Thisismydigitalself, 2018-01-29:
Here's a short clip.
Let me know what am I doing wrong.

https://send.firefox.com/download/d9015bbfc8/#hs4FoE-T6UQRS66BrpdCig

jackha, 2018-01-30:
Ahh I think I know the problem here. In preferences under Opening and saving files there is "Default behavior when opening a project file". The first time it will ask (default behavior), but dependent on the choice this can become "Always import models". In that case the models are just imported and placed as it normally would. I would prefer "Always open as a project".

The reason for this preference is that you can also "save" (vs save as project) a 3mf file. In that case it doesn't contain Cura's settings.

Thisismydigitalself, 2018-01-30:
Sorry to kill the joy.
This is how i always keep my preferences.
Can it be that Cura has a bug and disregards these settings?

[image]

jackha, 2018-01-30:
Hmmm ok that's really weird... I really thought I nailed it ^^ (it exactly made that difference on my machine)

Can you send us the logs and the project file? Then we can have a look at it.

Thisismydigitalself, 2018-01-30:
I will do so shortly, Thank you.

Thisismydigitalself, 2018-01-30:
Went to look for Cura.log i stopped and checked my Cura.cfg file.
this is from [cura] section:

[cura]
categories_expanded = ;acceleration_layer_0;acceleration_print;acceleration_support;acceleration_support_interface;acceleration_wall;bottom_thickness;brim_width;jerk_layer_0;jerk_print;jerk_support;jerk_support_interface;jerk_wall;platform_adhesion;raft_acceleration;raft_fan_speed;raft_jerk;raft_speed;shell;speed;speed_layer_0;speed_print;speed_support;speed_support_interface;speed_wall;top_bottom_thickness;top_thickness;travel_compensate_overlapping_walls_enabled;wall_extruder_nr;wall_thickness
active_mode = 1
active_machine = LM
choice_on_open_project = open_as_project
dialog_on_project_save = **False**

Can this be the culprit?
Checking..

Thisismydigitalself, 2018-01-30:
Here's my Cura.log file. i deleted it, started Cura and created the same steps as in the video so the log contains only that.

[cura.log](https://github.com/Ultimaker/Cura/files/1676962/cura.log)

joyfromprocess, 2018-01-30:
Just wanted to add that I am also facing the problem of saved projects not remembering model position (and numerous other settings, it seems). If I save a project file in 3.2 and then re-open, not only are all my models scattered but I notice stderr.log reports an issue with the project file:

`WARNING - 3MFReader.ThreeMFWorkspaceReader.preRead [350]: File C:/blablabla/blablabla.curaproject.3mf is not a valid workspace.`

For what it's worth, I have my option set to "Always open as a project"

fieldOfView, 2018-01-30:
The same error pops up in @Thisismydigitalself's log. Could either one of you post a project file to inspect?

joyfromprocess, 2018-01-30:
Here you go:

[10mm-slicing-layers.curaproject.zip](https://github.com/Ultimaker/Cura/files/1679426/10mm-slicing-layers.curaproject.zip)

Thisismydigitalself, 2018-01-31:
For what it's worth, I wasn't asked by my Cura (both 3.1 and 3.2) how I would I like to open @joyfromprocess  project file and the attached screenshot is telling us it was not opened as a project.

[image]

joyfromprocess, 2018-01-31:
Yep - it loaded the models in the wrong place and it doesn't look like it loaded any of the main print settings that were applied when the project was saved. Not sure if it's supposed to or not, but it didn't load the machine either (since the project was saved on a Cartesian and you loaded it onto a delta) 

Besides the models themselves, per-model settings are the only thing I see loading when I open a project file.

nallath, 2018-01-31:
It could well be that the multi-buildplate changes caused the saving of the 3mf to go wrong.

Thisismydigitalself, 2018-01-31:
> It could well be that the multi-buildplate changes caused the saving of the 3mf to go wrong.

Does 3.1 support Multi-Builtplate? projects issues are the same with 3.1

I did not know about this MultiBuildplate until you mentioned it.
I gave it a try. very sluggish response by Cura. what i did found out which I think is interesting was that when I saved a project and then reopned it in MultiBuildPlate mode and chose the "Save to all built plates" - For a split second Cura arranged the models on plate as they were saved, and after this split of a second (like a quick flash fast) Cura reset the model's position to the same problem as without multiplate mode.

Hope this helps.

jackha, 2018-02-01:
What are the settings of "Automatically drop models to build plate" and "Ensure models are kept apart"? They should both be turned off if you do not want any models moved.

@Thisismydigitalself multi build plate is introduced in 3.2. "Arrange to all build plates" rearranges everything to as many build plates as needed. Multi build plate + project files are not yet fully supported as the current active build plate is only saved. If you open a project file the current active build plate will be overwritten, along with all the settings. One of the reasons that it's in Experimental now.

@joyfromprocess is the project supposed to be like this? It seems to load fine. I wonder why "some computers" get the not a valid workspace error. In that case I think it falls back to re-arranging on load (which may be a bug itself, .. I'm going to investigate that).

[image]

Thisismydigitalself, 2018-02-01:
I have "Automatically drop models to build plate" and "Ensure models are kept apart" checked at all times. Are you saying for loading projects one should turn these OFF? - Makes no sense to me.
Projects should always load the very same complete Cura setup before saving the project.

fieldOfView, 2018-02-01:
Ah, this sparked something...

The block in the project has an infill_mesh setting attached to it. Models with an infill_mesh do not get a convex hull in Cura (since the 3.2 release; see #2758). However, when loading the project, my guess is that a convex hull is created for the block (and push apart is triggered) before the infill_mesh setting is restored.

The pushing apart should be inhibited until the project has finished loading, and all (per model) settings have been restored.

fieldOfView, 2018-02-01:
@Thisismydigitalself, a workaround to keep your sanity would be to group your objects before saving a project. Could you test if that results in the objects staying together properly?

joyfromprocess, 2018-02-01:
@jackha - I always have "Automatically drop models to build plate" and "Ensure models are kept apart" turned off. Your screenshot shows the proper model arrangement at the time I saved that test project. Very odd that it loads incorrectly for both @Thisismydigitalself and me.

@fieldOfView - I will try playing with model grouping and see if it makes any difference. Even if we figure out model positioning with this workaround, the projects still aren't loading the main print settings for @Thisismydigitalself and me. Note the difference in print settings between the two screenshots above.

Thisismydigitalself, 2018-02-01:
> @Thisismydigitalself, a workaround to keep your sanity would be to group your objects before saving a project. Could you test if that results in the objects staying together properly?

For what it's worth, - Loading a project with grouped objects did the trick but only for objects position.
Cura did not ask me how I would like to Open the project file nor it changed the settings as they were when saved.

joyfromprocess, 2018-02-01:
@fieldOfView - I just observed the exact same behavior as @Thisismydigitalself on my machine:

> Loading a project with grouped objects did the trick but only for objects position.
Cura did not ask me how I would like to open the project file nor it changed the settings as they were when saved.

Thisismydigitalself, 2018-02-01:
Guys, Check out how @joyfromprocess project opened up on my Cura and as it should after i deleted the content of C:\Users\User\AppData\Roaming\cura\3.2\ folder and created a new CR10 printer.

I can't remember when was the last time i got a project file to open up as it should. 
What a strange feeling..

[image]

joyfromprocess, 2018-02-01:
Interesting. Guess I need to consider wiping all my Cura settings and starting over. I had just done that after installing 3.2, but perhaps I annoyed the Cura gods when setting everything up again.

For what it's worth, that is better but still not completely correct. There should be multiple blocks stacked on top of each other in the center of the build plate, as seen in the earlier screenshot from @jackha

Thisismydigitalself, 2018-02-01:
Wait,.. there's more!

Thisismydigitalself, 2018-02-01:
After clearing 3.2 folder and checking with CR10 printer, i went ahead and recreated my Delta printer.
then tried loading the project again. I was happy to see that i get the proper message before openning the file. please note the PLA material as a reference to what comes next..
[image]

Then got the second confirmation message - Good.Material still set to PLA..
[image]

Now, the moment Cura opened the file - it switched from my Custom FDM Delta printer to MS2. Please note PLA changed to EDC and settings changed as well.
[image]

And here is where i manually switched printer to CR10:
[image]

And now to my Delta - Please note we're back to PLA and my default settings which are not relevant to the loaded project:
[image]

Thisismydigitalself, 2018-02-02:
I managed to screen capture my screen while Cura was loading the project and managed to catch a split second where the middle box is higher, much like @joyfromprocess and @jackha project opens up, but for only a split second, than the box drops in height.

[image]

Thisismydigitalself, 2018-02-06:
Cura 3.2 was just released so i was eager to try it out and see if saving/opening project issue was solved, just to find out it was not..

Then i did my thing and again deleted 3.2 folder content and started Cura 3.2 again. this time it booted up clean. then i tried loading previous saved project and like it should it asked me if I want to open it as a project. so guys - defiantly something is being copied from previous installed versions (I have 3.1 running side by side to 3.2) whch f**s things up. I will now check other projects uploaded to this thread and see how they are loaded.

Thisismydigitalself, 2018-02-06:
Tried uploading @joyfromprocess project:
[image]

After 3 seconds, Cura changed the cube by itself and this is the outcome:
[image]

I am so sorry to say that I can't work with Cura this way.
I really am sorry. I just can't trust what it generates.

👎

Thisismydigitalself, 2018-02-06:
Here's a video clip:
https://expirebox.com/download/1e328870c9ac44ed3f5a39de2c9ae192.html

jackha, 2018-02-07:
Deleting the 3.2 folder deletes the preferences as well. Are those settings turned on? Hopefully that explains what happened.

- Ensure models are kept apart  <-- this will push models apart
- Automatically drop models to the build plate  <-- this will drop every model on the build plate

jackha, 2018-02-07:
Turning off those 2 settings gives this result:

[image]

Thisismydigitalself, 2018-02-07:
Turning these off yield the same result on my end.
If opening Project does not override these settings then this is no good.
Please also note the settings - do you also get no top layers?

[image]
[image]

Thisismydigitalself, 2018-02-07:
Please be kind and tell us what's cooking and how Cura 3.3 beta is going to manage projects.
What one should be expecting to see and get in regards to the behavior of saving and opening projects.
Will we get to save and open project and get the exact same settings as they were when project was saved?

fieldOfView, 2018-02-07:
Sofar nothing has changed regarding the saving or loading of projects. I guess they're still examining what goes wrong exactly.

Thisismydigitalself, 2018-02-08:
A good start would be to label this issue as a bug.

fieldOfView, 2018-02-08:
A bug report has a bug description and clear steps to reproduce, this issue does not. Yes, we can read all of the replies and try to figure out what is the exact bug, but you really want to make it easy on developers to find the problem in the code. Not because developers are lazy, but because there are so many issues to keep track of, and developers are lazy. ;-)

Thisismydigitalself, 2018-02-08:
Reading all the comments and questions being asked it seems to me this is more like a combination of having no clear definition of what's a Project really is, how it should work and while digging in some related bugs surfaced. I am well aware of all of you guys hard work especially when done by lazy people :)  

My problem is seeing this issue across all latest 3.x versions and not really knowing what's cooking.
"luckily" i see more and more are having the same issue so i really hope this will be addressed sooner than later cause i save and load projects all the time, well - Less than I would have hoped for.

nallath, 2018-02-08:
There is a pretty clear distinction on what a project is;
- it contains all the info of a "standard" 3mf file (All models, all grouping, all positions, all per object settings).
- It also contains; Your machine, all active profiles; including materials, qualities, user settings, your setting visibilities.

Thisismydigitalself, 2018-02-08:
Well, Few posts above I was told that before i open a project I need to turn off:

- **Ensure  models are kept apart**  
- **Automatically drop models to the build plate** 

This is what i meant by saying I feel there is no clear definition.

fieldOfView, 2018-02-08:
@nallath: Does the project (try to) include the settings in cura.cfg?

nallath, 2018-02-08:
It saves the preferences; it only loads the visibility from them.

As for a clear distinction; I'm saying what is being **saved**. If you load a project, cura puts everything in the right spot. You can however, have certain settings that make cura do something *after* the project is loaded. If you have "drop models on the buildplate" enabled, cura can first load the model so that it's above the plate. It then sees that the model is above the plate and does what you asked it to do (eg; Drop the model).

So if you want an exact copy; you should disable those options.

Thisismydigitalself, 2018-02-08:
Hi, Thanks for this discussion.
To begin with i want to keep my global settings as I like them to be cause this is how i load stl's and work with them. secondly, even when i unchecked these i did not get the exact same settings @joyfromprocess claimed they were. When i save a project - i save my work, not a model. when i come back to it few days later i do not and should not trust my memory how Cura was configured. i should click open and that's it. If what I'm saying makes no sense and things are going to stay as they are then this is not really encouraging.

fieldOfView, 2018-02-08:
Even though I hardly use projects myself, so this is not much of an issue for me personally, I agree with @Thisismydigitalself that the current behavior is not expected with the term "project". The data is there, and loads as expected for a split second, only for the parts to fall apart. I can imagine that would be frustrating.

Thisismydigitalself, 2018-02-10:
@joyfromprocess  
Your experience with Cura is literally same as mine. 
One thing thou: if you'll look at the screenshots i've posted earlier you'll see your benchy came out without top layers. was that the way you saved your project? can you please double check and tell us if so, cause if not, it's more than adding these two global settings to Cura's project but there are still issues with visibility settings as well.

Thisismydigitalself, 2018-02-11:
Strange. Since yesterday while trying to open @joyfromprocess project, cura 3.2 crashes. it was not so few days back.

[cura.log](https://github.com/Ultimaker/Cura/files/1714211/cura.log)

Ghostkeeper, 2018-02-26:
I'm considering the original question in this thread as answered (by the first reply).

Let's not make this thread into a hotchpotch of all issues with project files together. If you find bugs with it, please make separate bug reports and include reproduce steps!

ianpaschal, 2018-02-27:
I'm reopening this and labelling it as a discussion at @Thisismydigitalself's request. It's true there are still parts which are unclear.

Ghostkeeper, 2018-03-01:
What is unclear, then?

Thisismydigitalself, 2018-03-01:
This is the problem with many and long threads. to summarize:  
I was told that before opennig a project one needs to turn off:

**-Ensure models are kept apart
-Automatically drop models to the build plate**

This makes no sense. Opening a project one should get the same settings as they were saved regardless how Cura is configured.

Ghostkeeper, 2018-03-01:
You get the same settings. Loading a project file will not get you the same preferences except for which settings are visible (and I think that's good; project files should not modify your language, sending anonymous data, etc). Whether you get the models in the same position remains to be seen.

Because the preference states that models have to be kept apart. If the project file was saved with models that are not apart (because the person who saved them didn't mind that they are too close) then Cura will move them apart for you. Likewise for automatically dropping files to the build plate.

We could choose to disregard this preference upon loading a project. But when you then move any object they will suddenly move apart again and drop to the build plate.

Thisismydigitalself, 2018-03-01:
Like we have this Keep/Discard pop-up window when we alter settings (And i said it before and will say it again: I always get confused by these questions and never really know what will happen. maybe it's only me) then we need the same (errrr..) pop-up window telling us if to keep/Discard:

-Ensure models are kept apart - Y/N ? (And how they were when saved)
-Automatically drop models to the build plate Y/N ? (And how they were when saved)

ianpaschal, 2018-03-01:
I've added the UX tag and will page @LukeChen-Ultimaker since I think to some degree this boils down to a UX question. Personally I've never been that bothered by loading projects because... well... I don't. I just re import my models and select saved quality profiles.

But it all depends on how the user expects Cura to behave and in that sense I would say it's maybe not as intuitive as it should be compared to more conventional software's total saving of everything (i.e. if I save a Word file in Word, I expect everything to be _exactly_ how I left it when I reopen it).

Thisismydigitalself, 2018-03-01:
"So let it be written, so let it be done" :)

FalloutBe, 2018-03-05:
I had the same issue. Opening a project file didn't open the saved slicing settings. Position of the object was also not saved.

Tried reinstalling: did not fix the problem.
I followed the instructions of Thisismydigitalself: deleting all the settings in the Roaming folder.
(I uninstalled Cura, manually deleted the roaming folders, and then reinstalled Cura)

This fixed the problem for me. Sadly my printer profiles are gone now, but at least I can now use cura..

fusentasticus, 2018-09-25:
RE: meaning of "project".... so in addition to the 
- couple of ignored settings/preferences when opening project (this thread)
- ignored path names (mentioned in this thread and issue still open as  https://github.com/Ultimaker/Cura/issues/3232)

I noticed that 
- in [3.4.1] when opening a project (both ways!) containing support blocks, it's not possible to delete a support block (Del key) without deleting the project model as well.

That's a third way (?) where a project doesn't behave as users would expect I think.  That is, as a user I expect the state to be restored.  For this particular issue see also my post https://github.com/Ultimaker/Cura/issues/4424 Can't save and load support blocks -- any workarounds?

fieldOfView, 2018-09-25:
> in [3.4.1] when opening a project (both ways!) containing support blocks, it's not possible to delete a support block (Del key) without deleting the project model as well.

Sure it is, in the same way as before you save and load the project:
* select the support blocker objects with ctrl-click and delete it, or:
* select the support blocker tool and click the support blocker object

fusentasticus, 2018-09-27:
Thanks fOV, my mistake --- it was the complementary action that I needed. 

My prototype workflow involves making 10 versions of a "thingy" until it's right. Thingy is complicated and can be printed only using Cura's amazing support interface with selective blocking. (Offsetting surfaces in the CAD program is way too complicated). 

So it's the action of deleting the project model, without deleting carefully constructed support blockers, that I'd need.  (That's what I meant to say is not possible currently.) Then I'd `File -> Open` to reload the model. (I'm using the CAD program to position the model, so that part of the  `Reload All Models` I don't need.)

I don't think this behavior is actually wrong by itself. What is unfortunate -- and that's the bug as smartavionics pointed out early in this thread is that https://github.com/Ultimaker/Cura/issues/3201#issuecomment-361200411

> When loading a project file, you should get the complete state.

is not true. The most important part of a project is that it's ongoing -- and Cura's project concept does not AFAIK support a project with evolving model design. I think that's a bit puzzling because tinkering with a complicated model would be characteristic of some industrial FDM use.  The more good stuff you put inside Cura (and that can't be accessed from the outside as in my closed comment https://github.com/Ultimaker/Cura/issues/4424), the more important the reload issue (incomplete state) of this thread will become.

[3 more replies]

closed 2026-09-08 by GregValiant