Ultimaker/Cura #3202: Slicing leaves exposed skin

danilius, 2018-01-28:
**Application Version**

3.1.0

**Platform**

Windows 10

**Qt**

N/A

**PyQt**

N/A

**Display Driver**

nVidia 23.21.13.9065

**Steps to Reproduce**
Load the attached model. Slice with 0.2mm layers. Run up and down the layers, there will be an exposed area at layer 41 that should have been skinned.

**Actual Results**
Slicing is incorrect, an area of surface is left exposed with infill, beneath an overhang.

**Expected results**
No exposed surfaces

**Additional Information**
[image]

[casting case back.zip](https://github.com/Ultimaker/Cura/files/1671560/casting.case.back.zip)

smartavionics, 2018-01-28:
Hi, it slices OK for me using the current master branch version. Layer height & first layer height are 0.2mm.
I've played with various settings and can't reproduce what you have above, sorry.

[image]

danilius, 2018-01-28:
If I use the standard settings, everything works fine, but using one of my custom profiles causes this, and I cannot track it down to any specific feature.

jackha, 2018-01-29:
@danilius You could have a look at the Skin Removal Width and Skin Expand Distance. See what happens if you set them lower or to 0.

danilius, 2018-01-29:
@jackha Bingo! That worked. I changed the Skin Removal Width to 0mm and the Skin Expand Distance to 1.2mm. Interestingly, there seems to be some other setting conflicting with this, since when I set up a new profile based on default settings, this problem went away.

closed 2018-01-29 by jackha