Ultimaker/Cura #3204: Feature Request: Expert Mode ti get rid of all the annoying warning

Spirou42, 2018-01-29:
While Cura and the UM3  has a very safe setup to assist new and inexperienced users the uttered warnings are annoying to an expert. 

I really would appreciate an "expert mode". Such a mode should - in my opinion - disable all warnings in Cura and also on the UM3 so that i don't have to acknowledge any slight and wanted discrepancy to your very safe setup. The flag could be secured by a method like github uses to prevent deletion of a repository.

nallath, 2018-01-29:
What kind of warnings would you no longer like to see? The colors?

Spirou42, 2018-01-29:
exactly. also the messages about not supported material/PrintCore settings in Cura connect on the UM3 which go along with custom materials.

Spirou42, 2018-01-30:
You can think of this request as a follow up to the late discussions on #3088. 
While the main problem of #3088 is solved in 3.2b. Profiles for custom materials are working again, a strange oddity remains. The current handling of the material type leads to and annoying situation which can't be fixed easily. 

So at the moment i'm in a situation where i can choose between two situations: 

1.  I can use custom material types
pro: easy access to the material in the ui
cons:  orange colours, compatibility warnings, the print job has to be approved on the UM3.

2. I stick to the given material types
pro: no warnings, no approval - unless I have the Soldex core installed (i'm using a Soldex with ruby nozzle for the metal filled filaments and for )   
cons: Materials get messed in the UI. 

So while  the current handling of the material type property can't be changed to resolve this issue I came up with this idea of a even more convenient solution (from my perspective). I'm an experienced user,  and I get annoyed if some piece of software tries to look after me and bothers me with dialog boxes or messages telling me stuff I know for centuries.

Ghostkeeper, 2018-02-14:
We've sort of been assuming that if you're experienced enough to modify your printer in a meaningful way, such as replacing the nozzle with a custom one, then you'd also be experienced enough to add your nozzle profile (variant) to Cura as well.

Spirou42, 2018-02-14:
@Ghostkeeper I would love to have the possibility but I haven't found a way to add nozzle variations  or core variants in Cura other than to edit internal configuration files. I could perhaps try to fully understand the way Cura handles its zoo of configuration files snippets but I wan't to use the printer as a tool not as a project. 
In addition Curas configuration file handling is way to fragile to simply try. A single mistake even a typo can prevent Cura from starting all together - hanging while "loading machines". I already have wasted a couple of days down there and even if I would have been successfully adding a nozzle description it would not remove the warnings and messages for custom materials (#3088).

Ghostkeeper, 2018-02-21:
In my experience, modifying configuration files wrongly will normally result in that specific file not being loaded and an error in the log regarding it, or result in that specific file being loaded but not matched with the current configuration (the latter can be confusing).

Spirou42, 2018-02-21:
@Ghostkeeper: I might have come to this conclusion (fragile config file handling) while trying to get custom materials and profiles working again. Issue #3088 had a special side effect - the last saved profile was loaded after program restart. So i had a simple script that renamed profiles in ~/Library/Application Support/cura and copied some files around so i could at leats change profiles by stoping cura , renaming/copying profiles, restarting cura.

While playing around with the profile settings and names, i encountered different situations in which cura won't start any more. Some of the situations where non reversible, Only cleaning the "~/Library/Application Support/cura folder help reproducibly. 

Normally i wouldn't complain. I'm a developer myself and i consider this stuff as "application private" otherwise there would be UI or a documentation for those situation. So tempering with configuration files directly is ok for but in that case i'm on my own.

Spirou42, 2018-02-21:
@Ghostkeeper: 
TL;DR: I think I found the cause of some of the crashes I encountered. **Broken References** to config files.

Have a look at [stderr.log](https://github.com/Ultimaker/Cura/files/1743992/stderr.log)

I had created a variant for my Soldex HardCore by copying a the most basic variant (_ultimaker3_aa04.inst.cfg_) from the application bundle. Changed name,author,line_width and machine_nozzle_id and and every thing seams fine until...
i copied the 0.4 variant into a 0.8 variant and renamed the 0.4 variant started cura321 again. 
OOPS, nice crash reporter you have implemented (Report send including this issue number)

Removing the copied variant did not remove the crash. Even cleaning the variant dictionary didn't helped.

I had renamed the copied _ultimaker3_aa04.inst.cfg_ into _ultimaker3_SoldexHC04.inst.cfg_. 
After checking that i can select the new core i renamed the 04 variant into _ultimaker3_SoldexHC0.4.inst.cfg_ cause all other files in the resources/variants directory of the application bundle used the size with a "." except that file i copied (_ultimaker3_aa04.inst.cfg_).

So a broken reference, in  this case the selected  _ultimaker3_SoldexHC04.inst.cfg_ from  _~/Library/Application Support/cura/3.2/extruders/ultimaker3_extruder_left+%232.extruder.cfg_ caused the crash.

I think a crash is a bit harsh. Maybe there is a sane default to fall back to. 

Carsten

Ghostkeeper, 2018-03-13:
Our "sane default" should be to ignore the file that contains the broken reference, so `ultimaker3_extruder_left+%232.extruder.cfg` would be ignored and subsequently your active machine instance too, causing Cura to pop up the dialogue to add a new printer, which would be instantiated with correct references again.

We're looking into this again (developers, see CURA-5045), since we made a refactor that put more sanity checks in which hard-crash Cura on purpose, which is not what should be happening.

Spirou42, 2018-03-14:
Cool