Ultimaker/Cura #3206: Reset calculated value does not always work

ChrisRiddell, 2018-01-30:
**Application Version**
3.2 Beta

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**
AMD 18.1.1

**Steps to Reproduce**
Change a setting that's normally calculated and click the ! icon next to the input field.

**Actual Results**
Nothing happens when clicking the ! icon to reset it back to its calculated value. It only happens on some settings not all it seems very hit and miss but errors are been logged.

**Expected results**
It to reset to its calculated value

**Additional Information**
[cura.log](https://github.com/Ultimaker/Cura/files/1676337/cura.log)

fieldOfView, 2018-01-30:
It can happen that the calculated value turns out to be the same value as the value that was there. Eg: you (or the profile) put the number 0.5 in a setting field, overriding the result of a calculation. The result of that calculation could still be 0.5. The only difference is that when the other settings that make up the calculated result change, the calculated result will change.

ChrisRiddell, 2018-01-30:
if I change say reg fan speed to 0 it will not let me change it back to the calculated value that is 20% on the ABS profile, I was going to export a cura project to show but for some reason its not letting me might be related?

updated cura.log

[cura.log](https://github.com/Ultimaker/Cura/files/1677052/cura.log)

ChrisRiddell, 2018-01-30:
A little background this profile was made by using the UM2+ ABS Normal, Changing settings and using the "Create profile from current settings/overrides..."

Ghostkeeper, 2018-02-21:
I tried with UM2+ ABS Normal, setting Regular Fan Speed to 0. When clicking the *i* icon it resets back to 20.0. Maybe it was fixed by now?

ChrisRiddell, 2018-02-21:
The reset to calc works until you save the profile.

Gif that shows it better
[image]

Ghostkeeper, 2018-03-13:
I think this is fixed then in the current master branch. We've rewritten some of the stuff that handles this in a [big refactor](https://github.com/Ultimaker/Cura/pull/3377) and I can't reproduce the issue after that.

ChrisRiddell, 2018-03-13:
Now I have Cura running from source latest from master branch I can confirm this has been fixed 😄

fieldOfView, 2018-03-14:
In my best Morpheus voice:
“Welcome... to the *real* world.”

ChrisRiddell, 2018-04-27:
@Ghostkeeper This is back in the official release of 3.3.1 Windows 10

changed fan speed and reg fan speed can change fan speed back by clicking the ! but not reg this is what the logs are showing 

> 2018-04-28 08:28:48,869 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:28:49,018 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [133]: Need to check for errors again. Discard the current progress and reschedule a check.
2018-04-28 08:28:49,133 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:28:49,771 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.6s
2018-04-28 08:28:59,008 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:28:59,597 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.6s
2018-04-28 08:29:00,745 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:01,462 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.7s
2018-04-28 08:29:05,180 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:05,794 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.6s
2018-04-28 08:29:07,526 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:07,672 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [133]: Need to check for errors again. Discard the current progress and reschedule a check.
2018-04-28 08:29:07,790 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:08,549 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.8s
2018-04-28 08:29:09,930 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:10,619 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.7s
2018-04-28 08:29:18,884 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:19,554 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.7s
2018-04-28 08:29:22,661 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:23,320 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.7s
2018-04-28 08:29:24,155 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [129]: New error check scheduled.
2018-04-28 08:29:24,798 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 0.6s

Ghostkeeper, 2018-05-25:
Can confirm. When I tried this with the Outer Wall Line with on an Ultimaker 3 it worked fine, but when I tried this with an Ultimaker 2+ and the fan speed it didn't. These were my reproduce steps:

* Add Ultimaker 2+.
* Switch to ABS.
* Set Regular Fan Speed to 0.
* Save this as a profile.
* Click the _i_ button next to Regular Fan Speed.

Actual results:

* Nothing happens.
* Nothing gets put in the log.

Expected results:

* The Regular Fan Speed setting resets to 20.0.
* An arrow appears to reset it to the profile value of 0.

diegopradogesto, 2018-07-03:
I have checked this last scenario in master. For UM3 works well but not for UM2+: Now the _i_ button has changed to _fx_ and it works when clicking it because the setting resets to 20, but then the arrow does not appear so we can never go back to the original value.

nallath, 2018-09-21:
Fixed this in master.

closed 2018-09-21 by nallath