Ultimaker/Cura #3208: Color picker for custom material opens behind the Preferences window (wrong Z order). Mac.

filipgoc, 2018-01-31:
**Application Version**

3.2 Beta

**Platform**

Mac OS Sierra

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

1. Add a custom material
1. click on the color patch
[image]

**Actual Results**

- the native color picker does open, but behind the preferences window
- this means that if the preferences window is in the middle of the screen (which is default), the color picker is completely invisible, and thus it looks like the function does not work. 

_in this image I moved the preferences window to show the picker hiding behind_
[image]

**Expected results**

The color picker should be on top of everything - so we can see it and pick a color 🌈 

**Additional Information**

ChrisTerBeke, 2018-01-31:
We had a similar report to this not too long ago but couldn't reproduce so far....

Vandrasc, 2018-01-31:
See CURA-2399, as it did not have a high priority it ended up in Won't do.

filipgoc, 2018-01-31:
Cool. As long as it's noted, we can see when and if it piles up. 
Feel free to close at your discretion.