Ultimaker/Cura #3209: Using Adaptive Layers Causes Cubic Subdivision to Slice Incorrectly

OtterofDeath, 2018-01-31:
[image]
[image]

**Application Version**
3.2.0

**Platform**
Windows 7

**Description**
When using both adaptive layer height and cubic subdivision infill, the infill is sliced incorrectly with the larger cells near the walls and the smaller cells in the center. This is opposite of what should happen with the denser infill towards the wall and the lighter infill in the center.

nallath, 2018-01-31:
@ChrisTerBeke

ChrisTerBeke, 2018-01-31:
We're aware adaptive layers doesn't work with some other features (that's why it's experimental). I'm not sure what the priority is of fixing those issues though.

Ghostkeeper, 2018-05-18:
Not a lot. We've pushed this off of our planning for now. We're going to have to do a big pass over the broken stuff in adaptive layers some day to make it reliable.

Firedrops, 2020-12-01:
Exploring infill patterns appears to have led me to this Issue.

I was using a 20x20x20cm cube with no features, nor fillets/chamfers whatsoever, and in Cura this was around 502 layers (I was using 0.24 mm layer thickness, and forgot that I had Adaptive Layers on).

I noticed that Cubic Subdivision density appears starts dense, reach its sparsest at ~layer 200, and then increased back in density at ~layer 319. It then decreased in density all the way until the top again.

To illustrate:
1st Infill layer
[image]

~layer 200
[image]

~layer 319
[image]

Last Infill layer
[image]

To verify, I turned off adaptive layers, and the infill density progression sliced as expected.

So this at least shows that if adaptive layers is turned on, even if it's not actually functional anywhere, it throws off infill generation.

Ghostkeeper, 2020-12-02:
Specifically it throws off Cubic Subdivision. Cubic Subdivision pre-computes the 3D cubes for the infill beforehand in a certain space that it thinks the model is taking up. Perhaps this pre-computed space is not big enough or it's sampling from that space at the wrong heights.

mr-september, 2022-09-12:
There's a certain irony when the 2 main features for 'adaptively' slicing the print don't work together 😂

SMH17, 2023-08-04:
I haven't checked the implementation, however considering that cubic subdivision infill seems to lose density in thin layer areas, I would guess cubic subdivision is computed on fixed layer height and when it tries to fill areas that have higher layer resolution because adaptive layer switch on thinner layers, it scales the geometry to fill the area. 
Both adaptive layers and Cubic Subdivision are very valuable features. It's a pity they still cannot work together.