Ultimaker/Cura #3214: 3.2 Beta Real state for printing is smaller then in 3.1

wkaissl, 2018-01-31:
**3.2 Beta**

**Windows 10/Surface Pro Ultimaker 3**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

jackha, 2018-02-01:
Please explain. 
The real estate is influenced by build plate adhesion brim settings, as well as the extruders used. You could check that first.

wkaissl, 2018-02-06:
Hello

Thanks for the information.

However the settings are exactly the same in both version and in 3.1 the real estate is big enough. Actually you can see on the grey shadows which I assume is the greyed out area that it is smaller in 3.2 Beta.

Best Regards

Wolfgang

jackha, 2018-02-06:
What machine did you choose? With specific settings?

wkaissl, 2018-02-06:
Ultimaker 3. Not that I’m aware of.

Best Regards
Wolfgang

jackha, 2018-02-07:
Ok I've created screens of 3.1 and 3.2, they both look like this (the screen is taken from 3.2). Do you see something else? Can you post a screen maybe?

[image]

wkaissl, 2018-02-07:
The 3.2 Beta

And here the 3.1

 

As you can see without “Generate Support” it works the same

But with “Generate Support” the 3.2 Beta fails and the real state is smaller.

Best Regards

Wolfgang

 

 <https://user-images.githubusercontent.com/1016604/35904983-8ad876e6-0be5-11e8-8a94-80e950230ca6.png>

fieldOfView, 2018-02-07:
Attachments send through email don't work. Please use the github.com interface to post the images.

jackha, 2018-02-08:
Ahhhh how is the support horizontal expansion setting? It is by default 0.2mm for PVA.

Ghostkeeper, 2018-03-02:
PVA has a pretty big horizontal expansion which forces you to put your models further from the edge. Another possibility is that the support was printed with the same extruder as the model in 3.1 but with a different extruder in 3.2, which cuts some volume from the left side due to the nozzle offset.

Both of these will have been the same between 3.1 and 3.2, but maybe you had different settings in 3.1.

Ghostkeeper, 2018-03-02:
I'll close this one because we could not reproduce this. Please take a look at whether the settings are exactly the same.

closed 2018-03-02 by Ghostkeeper