Ultimaker/Cura #3215: python error thrown after removing unneeded printer profiles

cjbeckjr, 2018-02-01:
**Application Version**
Cura 3.2.0

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Delete custom FDM printer profile

**Actual Results**
Error on loading

**Expected results**
application loads and runs

**Additional Information**

LipuFei, 2018-02-01:
Hi @cjbeckjr , could you share your log file? It is `C:\Users\<your-username>\AppData\Roaming\cura\3.2\cura.log`. And could you also describe in more detail which steps had you taken when that happened? Thanks.

cjbeckjr, 2018-02-01:
[cura.log](https://github.com/Ultimaker/Cura/files/1685326/cura.log)
Yes it shows trying to load non-existing, why is this loading from drive D? I will also remove all these files in the appdata as this it possibly why it won't load again

Ghostkeeper, 2018-02-26:
It shows that it's loading from drive `D:` and for later Cura versions from drive `X:` because those are paths encoded in the executable for Windows. On our build server the files are on the drive `X:` (previously drive `D:`). These .py files are packed inside the Cura executable though and so they don't have any path any more. The path that you get in your log is when it was 'frozen' by cx_freeze during our build.

The relevant error from the log is this:

    2018-02-01 00:19:18,802 - DEBUG - UM.Settings.InstanceContainer.getConfigurationTypeFromSerialized [465]: Could not get configuration type: Missing the required property 'definition'
    2018-02-01 00:19:18,818 - DEBUG - UM.Settings.InstanceContainer.getConfigurationTypeFromSerialized [465]: Could not get configuration type: Missing the required property 'definition'
    2018-02-01 00:19:19,209 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught error has occurred!
    2018-02-01 00:19:19,209 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
    2018-02-01 00:19:19,209 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
    2018-02-01 00:19:19,209 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "__startup__.py", line 12, in <module>
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "Console.py", line 24, in <module>
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\bin\cura_app.py", line 97, in <module>
    2018-02-01 00:19:19,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Application.py", line 338, in getInstance
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 311, in __init__
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 311, in loadAllMetadata
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 311, in __performEmit
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "C:\Program Files\Beta Cura 3.2\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 935, in _fillMissingVariants
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:     variants = registry.findContainersMetadata(definition = variant_metadata["definition"], name = variant_metadata["variant"])
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 229, in findContainersMetadata
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerQuery.py", line 78, in execute
    2018-02-01 00:19:19,240 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerQuery.py", line 72, in <lambda>
    2018-02-01 00:19:19,256 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "X:\3.2\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerQuery.py", line 107, in _matchString
    2018-02-01 00:19:19,256 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: argument of type 'NoneType' is not iterable

This looks familiar (the error in the `_matchString` function). It seems like some metadata of your container was suddenly `None`. I don't know where I've seen it before though.

I'm not able to reproduce your bug myself though, so I can't really debug it. These were my steps to reproduce:

# Add an Ultimaker 2+ printer.
# Add a Custom FDM Printer with default settings.
# Restart Cura.
# Remove the Custom FDM Printer. Since it was the active printer, you'll now switch back to the Ultimaker 2+.
# Restart Cura. Cura now launches fine.

Could you expand your reproduce steps, maybe? Was it the printer that you deleted or a material profile or a custom (quality) profile?

cjbeckjr, 2018-02-26:
I was able to reproduce the issue by deleting a printer profile in 3.2.0, I
have not tested this with the 3.2.1 update.
As many users had seen the similar issue I had them move files back to the
original file locations and this allowed Cura to load.

Looking at the changes in 3.2.1 this issue appears to be addressed.

ianpaschal, 2018-02-27:
> Looking at the changes in 3.2.1 this issue appears to be addressed.

Does that mean you're no longer experiencing it on 3.2.1? If so, I will mark this as `Fixed` and close the issue.

closed 2018-02-27 by ianpaschal

cjbeckjr, 2018-02-27:
Correct, in 3.2.1 the issue is no longer happening.