Ultimaker/Cura #3220: Cannot upload GCODE to ocotprint

NeoTrace82, 2018-02-01:
**Application Version**
3.2

**Platform**
Windows 10

**Steps to Reproduce**
Slicing the model works and saving it to the Harddrive, but i cannot get it to upload to Octoprint.
It did before with 3.1 and the API is correct.
Even the Message appears that it does upload for a second but dissapears straight away and nothing happens, as i would have pressed it at all-

NeoTrace82, 2018-02-01:
Dunno what happened but i tried to do it one more time to make a Report (Log/Screenshot) and now it worked... so i guess it can be closed

NeoTrace82, 2018-02-12:
It only workes time by time, most of the time when i start up octoprint and the Printer.
Seems like as soon as i e.g. preheat it cannot send any files to the Printer out of CURA.

fieldOfView, 2018-02-12:
Logs are needed. Go to Help -> Show configuration folder, locate cura.log and post it somewhere I can download it.

NeoTrace82, 2018-02-12:
018-02-12 08:54:29,368 - DEBUG - UM.Backend.Backend._logSocketState [181]: Socket state changed to Closing
2018-02-12 08:54:29,375 - DEBUG - UM.Backend.Backend._onSocketError [203]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2018-02-12 08:54:29,382 - DEBUG - UM.Backend.Backend._logSocketState [183]: Socket state changed to Closed
2018-02-12 08:54:29,405 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [722]: Backend quit with return code 0. Resetting process and socket.
2018-02-12 08:54:33,683 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2018-02-12 08:54:34,301 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [141]: Trying to access OctoPrint instance at http://192.168.0.148:50/ with the provided API key.
2018-02-12 08:54:34,488 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFinished [291]: API key accepted by OctoPrint.
2018-02-12 08:54:34,961 - WARNING - UM.Qt.QtApplication.__onQmlWarning [269]: file:///C:/Users/nTe/AppData/Roaming/cura/3.2/plugins/OctoPrintPlugin/OctoPrintPlugin/DiscoverOctoPrintAction.qml:236: TypeError: Cannot read property 'baseURL' of undefined
2018-02-12 08:54:35,411 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction.testApiKey [141]: Trying to access OctoPrint instance at http://192.168.0.148:50/ with the provided API key.
2018-02-12 08:54:35,620 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction._onRequestFinished [291]: API key accepted by OctoPrint.
2018-02-12 08:54:40,755 - WARNING - UM.Qt.QtApplication.__onQmlWarning [269]: file:///C:/Program Files/Ultimaker Cura 3.2/resources/qml/Preferences/MachinesPage.qml:164: TypeError: Cannot read property 'connectionText' of null
2018-02-12 08:54:40,778 - DEBUG - OctoPrintPlugin.OctoPrintOutputDevice.connect [333]: Connection with instance 192.168.0.148 with url http://192.168.0.148:50/ started
2018-02-12 08:54:40,794 - DEBUG - OctoPrintPlugin.OctoPrintOutputDevice.connect [333]: Connection with instance 192.168.0.148 with url http://192.168.0.148:50/ started
2018-02-12 08:54:40,845 - INFO - UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device 192.168.0.148 already added
2018-02-12 08:54:40,846 - INFO - UM.OutputDevice.OutputDeviceManager.addOutputDevice [103]: Output Device 192.168.0.148 already added
2018-02-12 08:54:41,052 - DEBUG - OctoPrintPlugin.OctoPrintOutputDevice._onRequestFinished [751]: Set OctoPrint camera url to http://192.168.0.148:50/webcam/?action=stream
2018-02-12 08:54:44,561 - DEBUG - SliceInfoPlugin.SliceInfo._onWriteStarted [66]: 'info/send_slice_info' is turned off.
2018-02-12 08:54:44,565 - ERROR - OctoPrintPlugin.OctoPrintOutputDevice._startPrint [479]: An exception occurred in network connection: name 'global_container_stack' is not defined
2018-02-12 08:54:46,564 - DEBUG - SliceInfoPlugin.SliceInfo._onWriteStarted [66]: 'info/send_slice_info' is turned off.
2018-02-12 08:54:46,567 - ERROR - OctoPrintPlugin.OctoPrintOutputDevice._startPrint [479]: An exception occurred in network connection: name 'global_container_stack' is not defined
2018-02-12 08:54:47,048 - DEBUG - SliceInfoPlugin.SliceInfo._onWriteStarted [66]: 'info/send_slice_info' is turned off.
2018-02-12 08:54:47,051 - ERROR - OctoPrintPlugin.OctoPrintOutputDevice._startPrint [479]: An exception occurred in network connection: name 'global_container_stack' is not defined
2018-02-12 08:54:47,209 - DEBUG - SliceInfoPlugin.SliceInfo._onWriteStarted [66]: 'info/send_slice_info' is turned off.
2018-02-12 08:54:47,212 - ERROR - OctoPrintPlugin.OctoPrintOutputDevice._startPrint [479]: An exception occurred in network connection: name 'global_container_stack' is not defined
2018-02-12 08:54:47,357 - DEBUG - SliceInfoPlugin.SliceInfo._onWriteStarted [66]: 'info/send_slice_info' is turned off.
2018-02-12 08:54:47,360 - ERROR - OctoPrintPlugin.OctoPrintOutputDevice._startPrint [479]: An exception occurred in network connection: name 'global_container_stack' is not defined

fieldOfView, 2018-02-12:
Thanks, I'll look in to it. To get the obvious out of the way, did you install the plugin through the Plugin Browser? Could you check if an update is available through the Plugin Browser?

NeoTrace82, 2018-02-12:
I did a clean install of CURA 3.2Beta so the plugin came with that.

Checked Plugins Browser it doenst say anything about an update
And the Version is 3.2.1

fieldOfView, 2018-02-12:
Please install the non-beta version of Cura 3.2.0. An updated version of OctoPrintPlugin is available for that version of Cura (plugin version 3.2.2).

NeoTrace82, 2018-02-12:
Just to be sure and give you some more informations it did work with previous versions 3.1 and below
and i have changed my Port for octoprint 192.168.0.148:50/

i can Monitor the Printer and see a Webcam stream

NeoTrace82, 2018-02-12:
did install 3.2 stable and upgraded the plugin, it is working now
Thanks for your help

very much appreciated!

fieldOfView, 2018-02-12:
Ok, thanks for confirming.
Can you close the issue again?

closed 2018-02-12 by NeoTrace82