Ultimaker/Cura #3221: Cura is slow to draw interface

eagleapex, 2018-02-01:
**Application Version**
3.1.0

**Platform**
Windows 10

**Display Driver**
Intel HD graphics 530

**Steps to Reproduce**
Open custom print settings to change values. Expand a list of options, such as Build Plate Adhesion

**Actual Results**
Windows or Cura
Draws
The
Items
one
every
two seconds.

**Expected results**
Entire UI should be drawn onscreen faster than I can blink. Most UIs are like this.

**Additional Information**
Most od Cura is slow to load and interface with. This program acting slower than most on my system.

dotdog20, 2018-02-01:
I'm having the same thing, tried 3.1 and 3.2.

**Platform**
Windows 10

**Display Driver**
390.65

ChrisRiddell, 2018-02-01:
Sounds similar to #3189

monouser7dig, 2018-02-02:
Changing values like layer thickness and so on is extremely sluggish on a recent mac quad core as well (Cure 3.1).

StefanBruens, 2018-02-04:
Also #3198

dagnall53, 2018-02-12:
sorry, I have same issue, interface is almost too slow to be usable.

dagnall53, 2018-02-13:
what version qt did 3.1 use, as that was fine!.

Ghostkeeper, 2018-02-21:
Cura 3.1 used the same Qt version as Cura 3.2: Qt 5.9.1. Do you mean that it got (significantly) worse between 3.1 and 3.2? Because I haven't seen that.

However, our project manager removed this bug from our planning so it's unlikely that we'll get to spend time to fix it. Cura's interface will remain slow to react for the unforeseeable future.

ianpaschal, 2018-02-21:
@Ghostkeeper Really? I was using 5.9.x at first and it caused lots of problems since I was under the impression we were using 5.8.2 and avoiding 5.9.x because bugs.

Ghostkeeper, 2018-02-21:
Oh, I thought we were on 5.9.1 because 5.9.2 was broken. 5.9.3 fixed that bug but we hadn't switched yet.

But I see that the upgrade to 5.10 failed because there were some styling issues. Apparently they switched back to 5.8, so I probably misremembered then. I'm on 5.9.1 myself.

ianpaschal, 2018-02-21:
Yes I am running 5.10.0 as the styling issues were only visible in built binaries so I'm not going to bother switching back.

But yeah, no distribution of Cura uses newer than 5.8.2 afaik.

Ghostkeeper, 2018-09-18:
We made a major improvement on this in 3.5. It's not perfect yet though.

kkriegel, 2018-10-04:
I came here after searching for "cura slow on Mac OS", due to my experiencing this issue with the latest version of Cura 3.5. I too found the UI update/response/render rate to be unusably slow.

This problem still exists on Mac OS X High Sierra, in the latest Cura version 3.5 downloaded from the ultimaker web site, which I downloaded on October 3, 2018. MacBook Pro (17-inch, Mid 2010), 8G RAM, NVIDIA GeForce GT 330M 512 MB. That's an early Intel Core i7 with 2 physical/4 logical cores.

@Ian
I'm curious (and I mean no offense): Does "This isn't really a measurable thing to test or fix" mean that you feel the problem description is too vague/subjective (hard to reproduce), or that you don't have access to a Mac OS machine to test on, or something else entirely?

ianpaschal, 2018-10-04:
@kkriegel It means both that:

1. It's hard to reproduce. On who's computer is it slow? What if it draws fast on my computer and slow on yours? What if it's speed depends on what other software you have running, etc.

2. There's no way to measure it being fixed. With a bug like "When you click it, it should do X but it does Y," you know it's fixed when "You click it, and it does X." In this case it does what it should, but should do it faster. Ok. How much faster? Does 10% speed improvement count as fixed? No? Should it be 50%? What if one part is fast but the others not, does that count as fixed?

That's what I mean. 😉

As I've said on other issues of this nature, Cura team tries to make speed improvements a constant goal. Sometimes there's something specific that we think about what is the most efficient way to do it, but in general we try to make everything as fast as possible so while we understand that it's frustrating for some people if the software runs slow, there's not a whole lot to do that we aren't already doing/striving to do.

kkriegel, 2018-10-04:
I guess I should say that on Mac OS, clicking on anything takes at least 10 seconds before the UI responds. That's any clickable object in any view/dialog, not just 3d rendering. On Windows 10, everything is perfectly fine. My solution will simply be to never try to run Cura on Mac OS, but run it instead on bare metal or a vm using Windows. That's because it works fine on Windows and the code base appears at first glance to be Windows-focused. I cloned the 3.5 branch out of curiosity.

Cheers,
Ken

fieldOfView, 2018-10-04:
> the code base appears at first glance to be Windows-focused.

What makes you say that?

ianpaschal, 2018-10-04:
> That's because it works fine on Windows and the code base appears at first glance to be Windows-focused. I cloned the 3.5 branch out of curiosity.

Somewhat ironically only two developers in Cura team use Windows. One (former, me) uses Mac, one uses Fedora, and the other four all use Ubuntu.

dickshaydle, 2019-11-15:
The settings menu is very slow. This reduces productivity for me because i have to wait when searching for settings.

### Application Version
4.3

### Steps to Reproduce
Unhide all the settings in the custom section of the sidebar
Search for a setting in the filter, i.e. 'ironing'. Remove search term.

### Expected results
all settings are shown in less than a second

### Actual Results
Ubuntu 18.4., Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz:    **15 seconds**
Arch Linux (everything latest version), X220 Notebook:   **49 seconds** !!!
Windows, X220 Notebook:   **24 seconds** 

Btw, also moving the mouse over the settings  brings CPU usage to 150% (1 core = 100%). But already idling uses 50% without a model and without the gui being shown? Thats only on Ubuntu 18.4.. On Windows and Arch there's only CPU usage <5% when idling.

Ghostkeeper, 2019-12-03:
We've made some improvements for that in Cura 4.4, Dickshaydle. Try it out!

kkriegel, 2019-12-04:
Ghostkeeper, et al:
I just tried 4.2.1 and 4.4.0 on Mac OS X. See my system specs above.

Nice and responsive UI in both versions with a ton of other software running at the same time (not a fair fight at all). Congratulations. This is now very usable on Mac OS X when compared to the 3.x versions. Well done!

dickshaydle, 2019-12-05:
I wouldn't go so far to call it responsive for my test, but it is a lot better now.

# Results for Cura 4.4
Ubuntu 18.4., Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz: **8 seconds**
Arch Linux (everything latest version), X220 Notebook: **20 seconds**

_so it is approx double as fast as in version 4.3_