Ultimaker/Cura #3222: Feature request: "round up" sizes of lines that would otherwise not get printed due to thinness.

MesteMonokrom, 2018-02-01:
There have been a few times where something hasn't been able to print properly because the lines have been too thin to slice.

An experimental setting that would round up any thin lines to the minimum extrusion size would be useful, I think.

Scaling up an entire print would leave it unnecessarily large, and completely redesigning a print because of a few lines that are a fraction of a millimetre too small for the slicer to choose to slice, isn't really ideal.

Thank you.

jackha, 2018-02-05:
Hi @MesteMonokrom ,
You could try turning on "Print Thin Walls", I think it does what you mean: it then prints pieces that are horizontally thinner than the nozzle size.

MesteMonokrom, 2018-02-05:
I can't believe I missed that setting. Will try it when I'm next at a computer, and close this issue meanwhile.

Thank you!

closed 2018-02-05 by MesteMonokrom