Ultimaker/Cura #3223: Cura inserts heating commands before custom start code

biscuitlad, 2018-02-02:
**Application Version**

2.7
**Platform**

Xenial Ubuntu

**Steps to Reproduce**

Add printer, K8400, add start and end gcode to heat and turn off left hand nozzle (necessary as can't create dual nozzle K8400 profile due to crashes).
**Actual Results**

Cura inserts:
```
;FLAVOR:Repetier
;TIME:3243
;Generated with Cura_SteamEngine master
M104 S190
M109 S190
```
_ before my gcode._ Therefore forcing me to wait until the default nozzle is heated, turned off, before I can start to heat my left nozzle. 
**Expected results**

Don't insert your own gcode - I want my gcode at the start - not yours!
**Additional Information**

ChrisTerBeke, 2018-02-02:
This is expected and by design. If you add your own temperature commands to the start gcode however we will omit the default ones.

closed 2018-02-02 by ChrisTerBeke

biscuitlad, 2018-02-02:
I do, and you don't.

biscuitlad, 2018-02-02:
Just tested it again - it still inserts those commands before mine. Tried inserting heating commands for the LH nozzle before everything else - it just ignores it.

fieldOfView, 2018-02-02:
You have to use the {material_print_temperature} (etc) replacement patterns.

ChrisTerBeke, 2018-02-02:
Code details: https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/StartSliceJob.py#L349.

Like @fieldOfView said, if you use temperature patterns in your own start G-code, we won't inject the default ones.

biscuitlad, 2018-02-02:
Nope, it's seems to be silent 'feature'. RepRap gcode style doesn't insert your gcode. Marlin & Repetier does. 

Is that by design?

fieldOfView, 2018-02-02:
Please post your start gcode, as you have entered it in Cura.

fieldOfView, 2018-02-02:
If your start gcode includes the string "{material_bed_temperature}" or "{material_bed_temperature_layer_0}" (without the quotes, but with the {}), the M190 for the bed heating will be omitted. If your start gcode includes the string "{material_print_temperature}" or "{material_print_temperature_layer_0}" (without the quotes, but with the {}), the M104/M109 for the bed heating will be omitted. Those are the specific string Cura looks for in the start gcode snippet.

biscuitlad, 2018-02-02:
OK, thanks. My comment above about the RepRap style gcode still holds - no insertion of any gcode from cura. 

Trying your suggestion with Marlin style, using the following gcode start, I get a random 190 inserted from the profile - should I follow {} with my own command or something?
```
;;cura only escape codes
{material_print_temperature}
M109 T1 S190.0
G28
G1 Z5 F600
G1 Z0 F600
M106 S75
G90
M82
```

I get 
```
;FLAVOR:Marlin
;TIME:3243
;Filament used: 1.66773m
;Layer height: 0.2
;Generated with Cura_SteamEngine master
;;cura only escape codes
190
M109 T1 S190.0
G28
G1 Z5 F600
G1 Z0 F600
```

karabas, 2018-02-06:
you need something like this
M109 T1 S{material_print_temperature}

lmctz, 2018-05-17:
Is there a workaround to prevent the temperature commands from being added, without having to put temperature commands in your gcode? I work with a printer that cannot read temperature commands in the gcode or it will crash, so the temp is set manually on the LED screen before every print. Cura automatically adds an M109 right after the ";Generated by" line. I usually remember to open up the gcode after generation and delete that line, but I'd love to prevent it.

fieldOfView, 2018-05-17:
Easiest way is to add this to the start gcode:
```
;{material_print_temperature} {material_bed_temperature}
```
This will add the temperatures in a comment, which is enough to convince Cura you have added your own heatup sequence.

lmctz, 2018-05-17:
fieldOfView, thank you so much, I'd been puzzling over this for a while.

jkexbx, 2020-07-15:
All my profiles were removed. I added new ender 3 profiles and they experience this issue. I just copied my old start up to my new profiles and it started inserting the gcode above mine. I even had every code it was inserting. Adding ;{material_print_temperature} {material_bed_temperature} causes cura to not insert junk above my gcode but frankly this isn't a solution just a workaround. Example:
[image]