Ultimaker/Cura #3224: Cura doesn't generate first layer raft for internal support if "brim only on outside" is enabled.

KubaAseph, 2018-02-03:
Cura 3.1.0 / Win10 x64

As I said in title. OFC it's kinda rational that the fishnet pattern won't print if it has nothing to connect to, but could you add some slicing alert about support lines mid-air? Or maybe let the net or support lines print on first layer then? For now no matter what they print only from second layer upwards, what results in instant spaghetti.
For me it ended only in broken thermistor wire, but at more advanced printers this could generate much more damage.

[image]
[image]
[image]

closed 2018-02-15 by KubaAseph

Ghostkeeper, 2018-03-08:
Support should always be printing on top of... something. Was support floating in mid-air?

KubaAseph, 2018-03-08:
Yes. The layer after the shown one was always support - both with inside brim enabled and disabled, but as you can guess on the 2nd one there was no first layer grid supporting the support so it was printed too high and scratched the rest of print off.

Ghostkeeper, 2018-04-03:
For me it always creates this fishnet pattern underneath all support. I don't see any support floating in mid-air.