Ultimaker/Cura #3226: Templates in start/end gcode use default profile value instead of current value

lumbric, 2018-02-03:
When adding variables (templates) to the start/end gcode, {infill_sparse_density} seems to be replaced with the default value of the current profile, not with the actual sparse density used for slicing.

**Application Version**
3.1

**Platform**
Linux Ubuntu 16.04.3 LT

**Steps to Reproduce**
Add  `;infill_sparse_density={infill_sparse_density}` to start gcode, select a slicing profile and change the value "Infill --> Infill density" without saving it to the profile. Then save the sliced object as gcode.

**Actual Results**
Gcode contains the line `;infill_sparse_density=20` where 20% is the default infill density of the chosen profile.

**Expected results**
Gcode should contain the line `;infill_sparse_density=50` where 50% is the chosen infill density, which is not stored anywhere.

**Additional Information**
It seems to work fine for some values (eg. layer_height), not for some others (infill_sparse_density, wall_thickness, ...?). 

The parameters which are not related to settings seem to work fine: ({day}, {date}, {time}, {print_time}, {filament_amount}, {filament_weight}, {filament_cost}, ...).

I found the list of available field names here:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

More related links:
* http://reprap.org/wiki/Start_GCode_routines
* #2041 and #798
* https://community.ultimaker.com/topic/11126-perfect-startgcode-endgcode-for-cura-15042-mac-for-octoprint/
* https://community.ultimaker.com/topic/6445-how-to-reset-startend-gcode/

I couldn't find any real documentation on this feature and implementation seems to have changed in the past in different Cura versions. Is this the expected result actually the desired one, or am I doing something wrong? Is there any documentation?

fieldOfView, 2018-02-03:
Could you try ```{infill_sparse_density,0}```?

In Cura 3.2, you can specify the extruder to get the setting from. I guess for infill_sparse_density the default value it takes is from the "global" stack, instead of the value from the extruder.

fieldOfView, 2018-02-03:
Yes, I am sorry, see my edit above.

lumbric, 2018-02-03:
> Edit: just saw that you are using Cura 3.1. This is a bug in Cura 3.1 that should be fixed in Cura 3.2
Ah ok, great -  thanks for the information!

If you don't plan to backport the fix to 3.1, I guess this issue can be closed.

lumbric, 2018-02-03:
Just tested with 3.2-beta, I can confirm that this works as expected. Both `{infill_sparse_density,0}` and `{infill_sparse_density}` return the current value.

lumbric, 2018-02-03:
Slightly off-topic: Is there any place where I can find all possible values? For example, is there any way to add the selected profile name?

closed 2018-02-03 by lumbric

Ghostkeeper, 2018-02-26:
Most of the possible values are listed here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json
But there are also some variables here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmextruder.def.json
And some metadata here: https://github.com/Ultimaker/Cura/blob/c16214e0f36c9bb80811c1ed19d458a0bef5b14c/plugins/CuraEngineBackend/StartSliceJob.py#L276

Getting the selected profile name is impossible though. You can only get setting values, the current time and the initial extruder number.

lumbric, 2018-03-18:
;SETTING_3 {"extruder_quality": ["[general]\\nversion = 2\\nname = Rubinstein Lo                                                                                              
;SETTING_3 w Quality\\ndefinition = kossel_mini\\n\\n[metadata]\\nsetting_versio                                                                                              
;SETTING_3 n = 4\\ntype = quality_changes\\nquality_type = low\\nextruder = fdme                                                                                              
;SETTING_3 xtruder\\n\\n[values]\\nbrim_width = 4\\ncool_min_layer_time = 10\\ni                                                                                              
80877 ;SETTING_3 nfill_sparse_density = 60\\nmaterial_print_temperature = 205\\nretrac                                                                                              
80878 ;SETTING_3 tion_amount = 0.3\\n\\n"], "global_quality": "[general]\\nversion = 2                                                                                              
80879 ;SETTING_3 \\nname = Rubinstein Low Quality\\ndefinition = kossel_mini\\n\\n[met                                                                                              
80880 ;SETTING_3 adata]\\nsetting_version = 4\\ntype = quality_changes\\nquality_type                                                                                               
80881 ;SETTING_3 = low\\n\\n[values]\\nadhesion_type = brim\\nmaterial_bed_temperature                                                                                              
80882 ;SETTING_3  = 65\\n\\n"}