Ultimaker/Cura #3228: Layer view hangs up with large file

tcostaphotos, 2018-02-03:
[engine overhang angles.docx](https://github.com/Ultimaker/Cura/files/1692240/engine.overhang.angles.docx)

**Application Version**
3.2 Beta

**Platform**
  Windows 

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
  loaded stl file made a few changes in printer support and then went to view in layers and system hangs up there.  Workpiece is greyed out.

**Actual Results**
   out

**Expected results**
 should be able to view project and see the layers and supports.

**Additional Information**

jackha, 2018-02-05:
What settings did you use that it doesn't show the layers? Possibly CuraEngine crashed. Is it possible to share a project file so we can reproduce it?

tcostaphotos, 2018-02-05:
Good morning Jack,

Thanks for your email.

Here is the file and the settings I used.

As I have been trying to work out some quality issues when printing this
engine block I have been keeping records on my printing settings.
My support angle is set for 87 due to the the fact that any setting lower
prints supports for the front of the block which is not needed and is not
technically touching the buildplate but it still produces it.
I tried emailing you the file but it is too large.
Here is a link to the file on Amazon.

https://www.amazon.ca/clouddrive/share/ytS4gJKoRatvsloGDxfXLAo3UMh6z9IqkWscDth0ZT9

Thank you again,

Tony Costa

Notes Block 11, Feb 3 2018, would not slice
Estimated Time 2d 18h 19m
Actual Time
QUALITY Cura 3.2
Layer Height 0.15
Initial layer height 0.2
Line width
Top/bottom line width 0.4
Initial layer line width

SHELL
Wall thickness 1.2
Wall line count
Top/bottom thickness 0.8
Top thickness 1.2
Top layers
Bottom thickness 1.2
Top/bottom pattern lines
Bottom pattern initial layer lines
Top/bottom line direction []
Fill gaps between walls everywhere
Ignore small Z gaps yes

INFILL
Infill density 30
Infill pattern lines
Infill overlap percentage
Gradual infill steps 0
Infill before walls yes
Minimum infill area
Skin expand distance 1.2

MATERIAL
Printing Temperature 203
Printing Temperature Initial layer 208
Build plate temperature 60
Build plate temperature intial layer 65
Diameter 1.75
Flow 100
Enable retraction yes
Retract at layer change no
Retraction distance 8
Retraction speed 65

SPEED
Print speed 55
Infill Speed 60
Wall speed 27.5
Outer  wall speed 30
Inner wall speed 60
Top/bottom speed 35
Travel speed 90
Initial layer speed 35
Initial layer print speed 35
Initial layer travel speed 40
Number of slower layers 2

TRAVEL
Combing mode All
Retract before outer wall unchecked
Avoid printing parts when traveling yes
Travel avoid distance 0.625
Layer start X 0
Layer start Y 0
Z hop when retracted unchecked

SUPPORT
Generate support yes
Support placement touching buildplate
Support overhang angle 87
Support pattern grid
Support density 35
Support line distance 2.2857
Support Z distance 0.1
Support X/Y distance 0.7
Minumun support X/Y distance 0.2

BUILD PLATE ADHESION
Build plate adhesion type Brim
Skirt line count 0
Skirt line distance 0
Brim Width 8

MESH FIXES
Union overlapping volumes yes

SPECIAL MODES
Print sequence All at once
Spirialize outer contour No

ChrisTerBeke, 2018-05-23:
No new info plus can't reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke

tcostaphotos, 2018-05-23:
Ok .  I can't reproduce with latest version of Cura either.