Ultimaker/Cura #3231: Wall Thickness Miss-Calculates passes for 0.8 nozzle

BloodBlight, 2018-02-04:
**Application Version**
3.1.0 (3.0.4 is NOT affected)

**Platform**
Debian 9.2

**Steps to Reproduce**
I changed my nozzle to a 0.8MM, changes my printer settings, but is it still pathing and extruding for a 0.4MM.

Wall count calculations also impacted (so if it is set for 1.6MM, it calculated for four walls rather than two).  It seems like the setting is not being completely honored when set manually on an original UM.  The change does show in the amount of filament used, so the setting is being processed to some extent.

Thorinair, 2018-02-05:
Could this possibly be related to #3197 ?

Ghostkeeper, 2018-02-27:
I'm getting the inverse of this.

Reproduce steps:
1. Add Ultimaker Original.
2. Load a model.
3. Go into Machine Settings.
4. Under the extruder, change Nozzle Size to 0.8mm. Close dialogue and let it slice.

Now this is what happens for me:
* The model's lines become much thicker.
* The line width setting remains the same.

Ghostkeeper, 2018-02-27:
Ah but it seems that my bug is only an update bug.

ianpaschal, 2018-07-31:
Duplicate of #3338 but fixed.

closed 2018-07-31 by ianpaschal