Ultimaker/Cura #3232: Feature request: save model file names in project so F5 works after project is loaded

smartavionics, 2018-02-05:
**Application Version**

Akk

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Load models onto build plate and save as project. Reopen project and then hit F5.

**Actual Results**

Models are not reloaded when  F5 is entered.

**Expected results**

Models in projects should be reloaded.
**Additional Information**

jackha, 2018-02-05:
This would be very good. Almost the same is also including the model names in the project file.

fieldOfView, 2018-02-05:
But should the models be reloaded or the project?

smartavionics, 2018-02-05:
F5 should reload the models irrespective of whether it has been saved/reloaded as a project. At the moment it really puts me off using infill meshes (or any other meshes) simply because if you want to change any of the meshes at all you have to delete them and load them afresh and then redo whatever positioning/scaling is required.

nallath, 2018-02-05:
This is pretty anoying to get right; A project also saves location & grouping of a model. So if you refresh, what should it then do? Only load the vertices again? Or should it re-do everything?

smartavionics, 2018-02-05:
I would expect F5 to have exactly the same effect when it is used before the project is saved and then again after it has been reloaded.

jackha, 2018-02-06:
@smartavionics I think that makes the most sense ^^. Then it doesn't matter if it's loaded from a project file or not.

Ghostkeeper, 2018-03-08:
Our project manager removed this bug from our planning because it wasn't seen as important. We won't be getting around to fixing it any time soon.

smartavionics, 2018-03-08:
Well, it may not been seen as important at Ultimaker but it's still damn annoying that it doesn't work.You really shouldn't close this issue simply because you haven't the time to fix it right now.

Ghostkeeper, 2018-03-08:
I'm not closing it. It should be clear that it's not yet solved but deferred until later. Don't hope too hard because we get a lot of bug reports incoming, but we might still get around to it or give more priority if enough ruckus is made.

brik2k, 2019-03-09:
FWIW, I'd like to add my vote to have this resolved. I have a number of saved projects that use multiples of the same model arranged in ways that make printing more efficient. If I make one tiny change to the original model I have to go through the entire process of multiplying, moving, and rotating all over again. If making Reload All Models work as expected is stretch maybe what about something like Replace Model where we could click on a particular instance of a model, select a new file, and have the new model loaded in the original models location and orientation. That would at least save some of the movement and rotation changes.  Thanks for considering. Love your work.

tturpin, 2019-09-11:
Hi,
Everything I could say has been discussed in this issue and #2124, I just want to add my voice to the request ! I never made the effort to search if including models by path was possible in Cura projects, until today, after wasting 3 hours printing the outdated version of my model, so I took the time and found this thread...

elf128, 2020-04-18:
I guess the issue is not resolved after 2 years. Hmmm.
Ok, please don't get me wrong. I'm not trying to put any pressure on core team. 
What is the real reason behind ignoring this issue? I'm sure there is good explanation.
Here is a list of my personal guesses, which might be wrong or/and unfinished.
  1) Team  is busy with other issues, while this one is not priority. 
  2) The issue has hidden blockers invisible from users point of view, which require re-factoring/re-implementing of critical components. It's time consuming and place some risks, therefor team focus on low-hanging fruits.
  3) Required implementation somehow conflict with other requests or global vision.

Can I hope that 3) is not the case and team will accept PR with the fix form someone outside of the team.

Thank you.

smartavionics, 2020-04-18:
Hi, this has not received any attention by the Cura devs but now, courtesy of fieldofview, there is a useful alternative in that the latest version of his MeshTools plugin for cura provides an option to replace selected models with a new model loaded from a file. Doing that preserves the position and transformations of the original model.

smartavionics, 2020-04-18:
Given that the MeshTools plugin now provides a perfectly good solution to this problem I am going to close this issue. Thanks for reminding me that it existed!

closed 2020-04-18 by smartavionics

Ghostkeeper, 2020-04-22:
Mostly 1) and a bit of 2) were the issue.

However we've made good headway into this issue, at least for reloading generic 3MF files. Some of it was provided initially by FieldOfView (also the author of the Mesh Tools plug-in) through a PR, though we found some more problems with it still. **We added that to the 4.6 release that released yesterday.** We do know one bug with it though that causes objects to be duplicated sometimes.

Reloading the profiles and such from 3MF is mostly 2) rather than 1).

Ghostkeeper, 2020-04-22:
See also https://github.com/Ultimaker/Cura/pull/7230 and https://github.com/Ultimaker/Cura/pull/7384. It's unfair to say that this hasn't gotten any attention from the Cura developers, but this thread number 3232 out of ~7500 was indeed not noticed any more.

ttait-vantim, 2020-10-15:
@smartavionics - can you elaborate on the mesh tools alternative?

I for one would sure still like this fixed. Once all the settings are arranged just so, its great to be able to save a project but then the model can never be updated... particularly annoying with multiple mesh modifiers. But if I don't save as a project, and cura is close then thats all lost.

How about about a "replace model" option in a project? Then one to could the select the model to replace, right-click and point to a new source file? That's actually cleaner than assuming the source files is in the same place as when the project was actually created.

smartavionics, 2020-10-15:
> @smartavionics - can you elaborate on the mesh tools alternative?

Install the mesh tools plugin from the market place.

> How about about a "replace model" option in a project? Then one to could the select the model to replace, right-click and point to a new source file? That's actually cleaner than assuming the source files is in the same place as when the project was actually created.

The mesh tools plugin gives you just that.

ttait-vantim, 2020-10-15:
@smartavionics - Thanks for the tip. I can replace the model, but the meshes no longer align and I can't find a way to make them do so...

The "replace model" option should really be part of the base program I think.

ttait-vantim, 2020-10-15:
Spoke too soon. It seems ungrouping, centering all models and re-merging brings them back into alignment.

spebf, 2020-11-23:
For those getting to this point. Until the issue is solved I use a workaround (in Cura 4.8.0). Just not saving the project and leaving Cura open. As soon there is an updated .stl available Cura ask if to update.