Ultimaker/Cura #3235: zero values of retraction in material print settings causes slicing forever

dandishonored, 2018-02-05:
**Application Version**

3.1 nightly build current

**Platform**

Linux

**Qt**

Qt 5.9.2

**PyQt**

pyqt 5.9.2

**Display Driver**

Nvidia 384.111-3

**Steps to Reproduce**

fill new material in "Manage Materials" and put zero values to "Print Settings" "Retraction Distance" and "Retraction Speed". Then let retraction active and try to prepare object for print (I did it for TPE material, where I don't use retraction and forgot retraction active)

**Actual Results**

when I clicked "Prepare" I got "Slicing" forever

**Expected results**

slice and generate G-code with retraction off

**Additional Information**

Just let user to turn off retraction in Material parameters for some custom materials. Great software ;-)

jackha, 2018-02-06:
You used 3.1 nightly build? Can you try it on 3.2 beta? Can you provide a project file that shows this problem?

dandishonored, 2018-02-06:
Behavior in 3.2 beta: it print warning that values are wrong, change "Prepare" button to grey(inactive) and set condition to "Unable to slice". User must know that values are in material preferences, otherwise search google for answer. I recreated behavior by "+Duplicate" and edit current PLA material, because "+Create" did not work for version 3.2 beta in AppImage format downloaded from Ultimaker homepage.
"+Create" button in 3.2 beta is doing nothing by clicking it. I added screen from Cura 3.2 beta where warning is visible. Better situation than in Cura 3.1 but still not ideal for newbies doing crazy things :-).
[image]

Ghostkeeper, 2018-02-27:
But it also colours the Retraction Retract Speed, Retraction Prime Speed and Retraction Speed settings red. You changed the retraction speed setting, but it has an error and it indicates that there is an error with the retraction speed, both with a message and with a colour on the setting. What more would you propose that we do to make it more clear?

dandishonored, 2018-02-28:
In 3.1 I can have Retractions Speed 0 mm/s and color is default black. I
can test it in beta again, but I'm using 3.1. I'm not sure if migrating to
beta is great idea, when I have all tuned in 3.1.

ChrisTerBeke, 2018-05-23:
Not a real issue it seems. Closing.

closed 2018-05-23 by ChrisTerBeke