Ultimaker/Cura #3237: Feature request: different wall and infill speeds on the first layer.

riodoro1, 2018-02-05:
**Application Version**
3.2.0 beta

**Platform**
macOS

It would be nice to specify different speeds for different moves during the first layer. I often find myself reducing the initial layer speed to very low values in order to better stick the fine details and later during print I increase the feed rate after they are printed so that the infill on large footprint parts does not take forever, then I need to remember to reduce the feed rate after the first layer so that the print does not fail. If I could configure let's say "initial layer wall speed" 10mm/s and "initial layer infill speed" 30mm/s the print time would be reduced and no supervision would be required.

fieldOfView, 2018-02-05:
There is no infill on the first layer, at least not unless you have specified to print 0 bottom layers.

riodoro1, 2018-02-05:
@fieldOfView I mean of course the solid infill on the first layer, does it have a special name?

smartavionics, 2018-02-05:
Hi, I have already use exactly what you are asking for and have already submitted PRs  so that it could be incorporated into mainstream Cura but it was rejected. See https://github.com/Ultimaker/Cura/pull/2356. It works really well and I use that capability on every job I print so I really do wish they would reconsider it.

ChrisRiddell, 2018-02-07:
I really like this and can see how it would really help with PETG.

smartavionics, 2018-02-07:
It works excellently with PETG, on my first layer I use 20 mm/s for walls and 40 mm/s for the skin and get perfect adhesion.

Ghostkeeper, 2018-03-02:
We need to have the collision resolved with the functionality of the "Initial Layer Speed" feature though. Basically this comment: https://github.com/Ultimaker/Cura/pull/2356#issuecomment-329495087

What do you think would be the best solution to allow Initial Layer Speed to keep working without introducing too many dials to turn?

rolandkohn, 2021-04-09:
Would really love to see this in the official build. Would help me very much with my designs that contain fine details on the initial layer.