Ultimaker/Cura #3238: Support within object perimeter lifts the object off the buildplate by one layer

iFreilicht, 2018-02-05:
**Application Version**
3.1.0
2.7.0

**Platform**
Windows 10

**Steps to Reproduce**
Load [this .curaproject.3mf](https://1drv.ms/u/s!AmgnMR2IYHpNk5QAPyJXL8HgiHw0vw) file with cura 3.1.0 and take a look at the first and second layer in layer view. I also tested this in Cura 2.7.0, and the same problems occur there.

**Actual Results**
You will notice that the first layer only contains the support structure and the skirt. The model itself starts in the second layer, which leads to the print failing spectacularly.

**Expected results**
The skirt should be around the model and the model should be on the buildplate.

**Additional Information**

Screenshots for Cura 2:
[image]
[image]

Screenshots for Cura 3:
[image]
[image]

iFreilicht, 2018-02-09:
I also tested this without any adhesion helper and with brim, but it doesn't change the behaviour. The manual workaround would be to remove the first layer in Gcode and modify all other ones accordingly, but that's quite tedious.

Ghostkeeper, 2018-03-05:
An easier workaround would be to move the object down in Cura by 0.051mm.

This reproduces for me. We've seen this sort of thing often when the bottom of a model isn't 100% flat. But at least I'd expect there to be support along the entirety of the first layer then.

Ghostkeeper, 2018-03-05:
Turns out this is more of a matter of Cura's front-end not being entirely accurate in its 3D convex hull calculation which results in not putting the model entirely flat on the build plate.

iFreilicht, 2018-03-05:
Just lowering the model by 0.051mm is a great workaround, thank you so much! It seems this problem also disappears for lower layer heights. If I set the layer height to 0.298 or lower, the model is placed on the buildplate properly. If it's 0.299 or more, it's raised by one layer.

diegopradogesto, 2018-07-04:
I will close this ticket since there is a workaround and maybe is provoked by a non perfectly flat surface in the model.

closed 2018-07-04 by diegopradogesto