Ultimaker/Cura #3239: Crash when extruder 2 is selected

nangad, 2018-02-06:
**Application Version**
3.2 Beta

**Platform**
Windows 10 Enterprise 64-bit

**Qt**
 from Ultimaker's website
**PyQt**
 from Ultimaker's website

**Display Driver**
Nvidia 390.77

**Steps to Reproduce**
Create Custom FDM with Number of Extruders set to 2 and Gcode Flavor set to Makerbot.
Add .STL file and set model to extruder 2.

**Actual Results**
During the slicing process a pop up appear with "CuraEngine.exe has stopped working"
Click on "Close the program" and Cura stay open

**Expected results**
the project should complete the slicing process

**Additional Information**
[Support Files.zip](https://github.com/Ultimaker/Cura/files/1697859/Support.Files.zip)

this only happen with Markerbot Gcode flavor.
tried multiple .STL files.

Ghostkeeper, 2018-02-06:
This also happens with the BFB flavour.

Ghostkeeper, 2018-02-06:
Relevant bit from the log:

    current_e_value = 0.0;
    extruder_attr[current_extruder].retraction_e_amount_at_e_start = extruder_attr[current_extruder].retraction_e_amount_current;

I'm getting this too.

Ghostkeeper, 2018-02-06:
No, I was wrong. I shouldn't disable the assertion. Instead I've let it reset the extrusion value for those flavours just like the rest. According to the RepRapWiki it's perfectly well supported, but according to the commit that introduced the exception it has some quirks on the firmware of those printers. Let's hope that the quirks are resolved.

It now outputs G92 E0 upon an extruder switch. I've still kept the exception that it won't use G92 after using >E10,000 (which is a tweak to improve accuracy if the firmware uses floats for their E). The extrusion values themselves look okay but you'll be the first to ever test multi-extrusion with Makerbot flavour via Cura, I think, so cross your fingers!

Ghostkeeper, 2018-02-27:
See https://github.com/Ultimaker/CuraEngine/commit/7abc629e2bf25255ac7b7c5141ee65570766932f

Did the 3.2 stable release fix it for you?

nangad, 2018-02-28:
Yes

closed 2018-02-28 by ianpaschal