Ultimaker/Cura #3242: travel not optimal while ironing

notmine642, 2018-02-06:
**Application Version**

cura 3.2.0 beta
**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

slice with ironing option
**Actual Results**

sometimes, travel make a strange path to go from one point to another when ironing
**Expected results**

it could be more strait forward, doing lines instead of complex path
**Additional Information**

smartavionics, 2018-02-06:
Could you please make the gcode available so that we can see what you are referring to? Thanks.

Ghostkeeper, 2018-03-01:
Creating the optimal path is [NP-hard](https://en.wikipedia.org/wiki/NP-hard), meaning that a (nondeterministic) quantum computer could find a solution in polynomial time. In practice, this means that it's exponential according to the best known algorithms to mankind so far. It would take extremely long to slice.

We have a new algorithm in 3.3 that could be used to make a more efficient path that almost always goes along the border. It still creates a scar, but much less frequently. However this new algorithm hasn't been written in such a way yet that it can be used for travel lines.

If you have a more specific problem then we can maybe give a more specific answer. We know that it's not optimal and it won't ever be completely optimal.

closed 2018-03-01 by Ghostkeeper