Ultimaker/Cura #3243: Dual Extrusion Issue when Switching Extruder.

nangad, 2018-02-06:
**Application Version**
3.2 Beta

**Platform**
Windows 10 Enterprise 64bit

**Qt**
From Website

**PyQt**
From Website

**Display Driver**
Nvidia 390.77

**Steps to Reproduce**
1. Create Custom FDM with Number of Extruders set to 2 and Gcode Flavor set to Marlin, Marlin(voluemetric), reprap, Griffin, Mach3, and Repetier. Can't test Makerbot or BFB.
2. Have two model with an extruder selected for each one. Ex. model 1 using material 1 and Model 2 using material 2.
3. save Gcode

**Actual Results**
When print switching extruder the wrong temperature is set. See OctoPrint Terminal log.txt for detail. Anything after "#" are my comments

Ex.
T1 
M109 S220
M104 T0 S175
...
M104 S230 
At the end this set T1 to 220 and T0 to 230

**Expected results**
T1 should be set to 230 and T0 set to 175

The code below work on my machine
T1 
M109 S220
M104 T0 S175
...
M104 T1 S230 

**Additional Information**
[Gcode extruder Switching issue.zip](https://github.com/Ultimaker/Cura/files/1700646/Gcode.extruder.Switching.issue.zip)

nangad, 2018-02-06:
I just notice 3.2 was release. I will try to reproduce this on the official release.

nangad, 2018-02-06:
3.2 produce the same Gcode as 3.2 beta and I notice the same issue with Makerbot and BFB Gcode flavor.

nangad, 2018-02-07:
It look like the CuraEngine gcodeExport.cpp @ line 1011 to 1033 determine if it the M104 command get the extruder T#. I think if you change line 1027 to "if (extruder != current_extruder || extruder_count>1)" should fix this issue.

Vandrasc, 2018-02-07:
@Ghostkeeper? Could you please have a look? Thanks!

@nangad, thank you for investigating and reporting it, we will have a look.

karabas, 2018-02-12:
I guess it's smoothieware based printer

nangad, 2018-02-18:
mighyboard based printer

nangad, 2018-03-04:
@Ghostkeeper Any updates?

Ghostkeeper, 2018-03-28:
As far as I know, the actual behaviour is correct here (not the expected behaviour).
See http://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature (and also see M109).

The `T1` command in the M104 code should specify the tool for which to set the temperature, not switch the tool itself for subsequent commands. That would imply that the `G1` commands are also executed by T1, wouldn't it?

I know that Smoothieware interprets these things different from the rest though. Maybe MightyBoard too?

Ghostkeeper, 2018-03-28:
Do we need to make an exception for Smoothieware here? I wouldn't know how to support MightyBoard unless by creating a new g-code flavour (or using Smoothieware flavour for those printers).

nangad, 2018-03-30:
I should explain my setup. the printer is PowerSpec 3D Pro 2(FlashForge creator pro clone) running Sailfish firmware.  I use Octoprint for converting the Gcode to X3G(S3G). I had the same issue with your and Octoprint  GPX convert app. 

looking at m109/m104 conversion from https://github.com/makerbot/s3g/blob/master/doc/GCodeProtocol.md showed that the s3q equivalent is  set_platform_temperature(tool_index, temperature). 

I'm guessing when tool_index is not define the code just uses the old value.

It seem like the best option is to copy the RepRap ( this flavor worked the best) and have M109/M104 always have the T parameters

jepler, 2019-01-01:
FWIW I believe that this is an issue in cura.  When using ONLY extruder 2, a line like "M109 S235" is output when using the reprap gcode dialect, with the intent of setting the temperature of extruder 2, selected by a "T1" gcode on some previous line:
```
;LAYER_COUNT:14
;LAYER:0
G1 F1200 E-9.5
T1
M82 ;absolute extrusion mode
M135 T1
M83 ;relative extrusion mode
M109 S235
M107
```

I only tested with ultimaker cura 3.4 so far, which I realize is not the current version.

jepler, 2019-01-01:
(this means that trying to "fix" things by postprocessing T0 onto any M104/M109 line which does not already have one doesn't work)

jepler, 2020-06-20:
Please clarify what the requested information is. This issue still affects me.

Ghostkeeper, 2020-06-30:
Stupid bot. I was still typing.

Looks like the requested information was a clarification on how Mightyboard parses g-code. Most printers interpret `M104 S200 T1` as "heat up the second extruder to 200 degrees", as per [this specification](https://marlinfw.org/docs/gcode/M104.html). There is one exception though: Smoothieware interprets it as "heat up the current extruder to 200 degrees, oh and also change the active tool to the second extruder".

Incidentally, we had a [contribution](https://github.com/Ultimaker/Cura/pull/7067) to support this part of their g-code flavour. So if it's the same for Mightyboard, Cura 4.7 will have a solution for you.

jepler, 2020-06-30:
I'll try 4.7 when it is available, and see whether the new dialect fixes anything.

closed 2020-07-14 by no-response[bot]