Ultimaker/Cura #3244: Cura 3.2beta inside brim printed in wrong order

lmftiv, 2018-02-06:
3.2 beta is printing the inside brims starting from the object wall. The inside brim should be printed the same way as the outside brim. Starting away from the object and ending at the object wall.

Now all adhesion and extrusion  issues will be at the beginning and at the wall of the object.

I have not tested this behavior on another versions of Cura.

Ghostkeeper, 2018-03-01:
Indeed, this has probably been going on for quite a while.

This is caused by the way that we plan these lines. We simply go to the closest brim line regardless of where we were. This is usually the outermost brim line because we usually start from the outside (except when using delta printers). However for inside brims the closest brim line is then the innermost line (closest to the wall).

hamster65, 2019-01-19:
I have this issue too with 3.6.0. I am trying to print an object with PETG filament but since the brim starts at the walls, it is not connected to the object. This results in failing prints because the corners warp upwards.