Ultimaker/Cura #3245: Retraction while performing combing travels please

ctzsnooze, 2018-02-06:
Sorry to bump this again, but when combing, having no retraction is causing blobbing that is a major issue for me.

It is particularly a problem when the travels are not over infill, but over thin walls.  What happens is that there can be multiple combing travels over those thin walls, each time with a tendency for gravity to permit tiny blobs to fall out the edge of the extruder.  With successive layers, these accumulate into gritty little sandpaper-like junk.  

With transparent, flexible filament like TPU, and with very thin walls, these blobs can accumulate badly, so that instead of having smooth walls, they get lumpy and uneven.

If the combing travel is over infill, these micro-blobs are of no consequence, getting wiped off on the infill.

But on models with thin walls and multiple combing travels, not being able to request retraction while combing is a major problem.

The easiest solution would be a 'retract while combing' option.  This could be a standard stop/retract retraction.   For short combing travels it's not needed, so perhaps the user could specify a distance after which combing travels are retracted.

A more sophisticated solution could be to only worry about doing a combing retraction if the combing travel goes along walls, - in other words, if the combing travel is entirely over infill, then don't worry about the retraction.

This has been proposed a long time back (#2531), and since then people with issues have grappled with possible solutions (e.g. #3077, #3119, #3077), but I'm not sure where we're at.  It's very important for me because I use blobby, retraction dependent clear filaments like TPU on thin wall hollow box type objects where both internal and external blobbing is an issue without combing.

I am confident that a retraction option for combing travels will be a big improvement and solve a lot of issues with combing.

Thanks and apologies for re-posting, just want clarity on what will happen about this.

PS there is a related issue that contributes to this problem, that being a convoluted combing path.  See https://www.youtube.com/watch?v=4snZRGQkU_Y  The problem was that the head combed to the nearest physical point, instead of the shortest combing path.  This resulted in multiple unnecessary combing travels.  With no retractions these caused a lot of blobbing.  @smartavionics had a proposed solution https://github.com/Ultimaker/CuraEngine/pull/617 but I don't know what happened to it. I'd love to know that this can be implemented asap.

smartavionics, 2018-02-06:
https://github.com/Ultimaker/CuraEngine/pull/617 is in 3.2. Cheers.

ChrisRiddell, 2018-02-07:
@smartavionics I don't think that made it to 3.2 as I see the merge commit in master but not 3.2 branch.

smartavionics, 2018-02-07:
Ah yes, you're right. Well it should be in 3.3, then.

ctzsnooze, 2018-02-07:
Great to know that Ultimaker/CuraEngine#617 will be in 3.3.

How about a retraction option while doing combing travels?  For me, this is super important.

smartavionics, 2018-02-07:
> How about a retraction option while doing combing travels? For me, this is super important.

I know a lot of people think that is what they need but I remain to be convinced. Personally, I think combing is pretty crap and more often than not, simply turn it off altogether.

ctzsnooze, 2018-02-08:
OK let me explain.

I make 'pods' for drones from TPU.  They are a kind of protective shape that encloses cameras etc inside.  Sort of a hollow object.  I need the interior and exterior walls to be smooth and free of blobs; both interior and exterior need to be dimensionally perfect.  I don't have time to be trimming little blobs or zits off afterwards (in some interior places it's kind of difficult to get to them).  The designs also have some very thin walls, say three layer wide (i.e. 1.2mm or so), and are organic, so the thickness varies (cannot be held at exactly 1.2mm).  So I need variable width lines to completely fill these walls as they gradually become thicker. 

I also need super strong layer adhesion, so I lean to over-extrusion to limit the amount of air in the walls and ensure all 'four' walls of the 'bead' of filament are touching each other and as strong as possible.  And TPU goes runny and is always a bit prone to forming small round blobs after long travels, even with retraction and coasting, as you well know if you've used it.

I feel that I have anti-blobbing optimised as much as possible - consistent with a solid strong shell.  

If I disable combing, I get lovely smooth thin walls.  Perfect.  But when the printhead travels across the open interior of the pod, from one side to the other, sometimes a little round blob dribbles out before it hits the other side and gets stuck there.  How bad this is depends on back pressure accumulation on previous lines, retraction settings, filament variation etc etc, but in reality I simply cannot prevent it absolutely, and it is very annoying to have to manually clip these little blobs away.  

If I enable combing, I get none of that interior travel blobbing at all.  The interior and exterior are - generally - perfectly smooth and blob free.  It's great!  Where the combing travels are over non-solid infill, the lack of retraction doesn't matter, the head de-pressurises into the air spaces in the infill, that's where these little blobs sort of vanish into.  If all my walls were wide enough to have non-solid infill, I'd have no problem with combing at all.  

But in reality, the walls along which the combing travels are actually very thin, so there is no air-space for the combing travels to de-pressurise into.  So the combing travels deposit a thin layer of dribble on top of whatever was underneath.  In this case it's a 3-layer wide thin shell.  It's easy to imagine that if there are like 5 combing travels over such a thin wall, each one oozing a bit because there's no retraction, then a bit of junk will accumulate and sort of blob up.  It actually gets sort of roughened and then accumulates by distorting the wall slightly; some excess goes outside the wall where it leads to a sort of fine sandpaper feel.  Instead of being glass-smooth, the walls become rough, like sandpaper, and just don't look good.

I am of the view that if the printhead did a retract before these thin-wall combing travels, then this sandpaper effect would be markedly reduced.  Maybe not completely, but quite a lot.  I say this because I know my retraction settings markedly improve blobbing in standard travels, so I would expect the de-pressurisation to markedly reduce this problem also. 

So, for me, combing is an absolutely fantastic feature that significantly improves my print quality. What I do is quite specialised, so it may not be for everyone; for 'solid' objects, with lots of non-solid infill, or with non-dribbly materials, combing might not be important - but for what I do it is a really big plus.

At the same time, I'm sure that combing would be even better - probably much better - if there was a option for retraction during combing travels.  

I'd be more than happy to test some code with a retract for combing travels option. Then we will know, rather than be guessing, whether the idea works.  I'm in a great position to test it out.

You may be right, it might not!  But if it did help my prints a lot, I'm sure others will find the same thing.

Sorry for making such a big song and dance about it, but I honestly anticipate it being a huge improvement.  

If there was a way to test by hacking the GCode to force a retract on *every* travel, I'd be happy to test it that way.

ctzsnooze, 2018-02-08:
First, combing on, second, combing off.
Note far fewer travels across the inner walls when combing is on.
BUT note also many, many un-retracted travels along the walls when combing is on.  These are what cause the problem:
[image]
[image]

ctzsnooze, 2018-02-08:
Here is where there are thin walls.  Again note that with combing on (first graphic) there are hardly any travels across the walls, meaning perfectly smooth inner walls.  In contrast, combing off means multiple travels over the walls.  But, if combing is on, there are between 5 and 8 un-retracted combing travels along the thin walls, they cause micro-blobbing and roughness.
[image]
[image]

smartavionics, 2018-02-08:
>In contrast, combing off means multiple travels that cross the inner walls, causing blobs that get stuck on the inside of the object, but no un-retracted travels within the walls, so the walls are smooth as silk, clear and transparent. Well, apart from the blobs.

Why are you getting blobs on the travels that cross inner walls given that retractions are in use? Where would those blobs be in the images above?

ctzsnooze, 2018-02-09:
I don't get a blob every time a travel crosses one of the inner perimeters, but they seem to be more frequent if the print head has been making solid walls and then needs to cross to the other side, across the interior.  Having been run over solid walls there can be a bit of back pressure buildup, and then despite retraction there can be some material that leaks out the nozzle tip as it crosses free space.  That material then makes a small blob which gets stuck on the opposite inner wall as the print head crosses to the other side.  With TPU, no retraction setting seems to entirely prevent this happening to some extent, and excessive retraction causes problems of its own.  These little interior blobs don't happen all the time on every crossing, just some of the time.  For instance, there will be a crossing from one bit of infill to another, that doesn't always result in a blob since there is no pressure buildup while doing infill.

Ghostkeeper, 2018-02-12:
What about a variation of Travel Avoid Distance that works on the inside of the perimeters as well?

Also keep in mind that retracting TPU is very detrimental to the print quality, since it stretches and compresses a lot in the path from the extruder wheel to the nozzle. It takes some time for the flow to stop and to start again.

ctzsnooze, 2018-02-13:
Ha ha yes TPU is annoying like that, however I find retraction and coasting to work well together and I am happy I've optimised those things as best I can.  

In my case yes I do ever so slightly over-extrude, I need to to get 'solid, strong' prints.  This means some nozzle back-pressure, particularly after long perimeter extrusion runs, which consequently are then a bit prone to blobbing.  But, even if I cut back the extrusion multiplier so I'm under-extruding, TPU sort of still just dribbles out the printhead on long travels, even with retraction - it's just runny annoying stuff.   And since its transparent, every internal air gap or under-filling or issue of any kind is very apparent.

I think that a lot of the fine sandpaper like blobbing on the thin walls is due to the combing travels being over already printed perimeters, where just a thin dribble can form tiny bits of junk either side of the printhead (sort of leaking out randomly on the edge of the nozzle) as it slides along over previously printed layers.  Do this multiple times and this junk then accumulates enough to create problems.  

One solution is to minimise combing travels generally, and I know that the zigzag coding and other improvements will help this.

However, I would think that a retraction during combing travels that went over perimeters is what I really need.

I guess if the Travel Avoid Distance worked on the inside of perimeters that might cause 'normal' travels to replace the combing travels.  However that would result in internal blobs as if there were no combing travels along the long thin walls in the above prints.  So, while I'm not sure, I doubt this would be the ideal solution.

I would be *so* grateful if these combing travels could optionally retract; the option would only be needed when:

- the combing travel was of meaningful length (i.e. the user should be able to set a combing travel distance below which there would be no retraction

- the combing travel would go over perimeter lines

As a result of the above settings, there probably wouldn't be many combing travel retractions required on most prints.

I would be *so* grateful for the opportunity to test this.

A standard retraction method should be OK.  I do accept that it can cause problems on its own but let's see?

Thanks, Ghostkeeper.

Ghostkeeper, 2018-03-07:
Have you tried disabling combing and then enabling the z-hop upon retraction? That should prevent the blob from attaching to the wall on the inside.

ctzsnooze, 2018-03-12:
Yes I have tried that, but then a 5 hour detailed technical print becomes almost a 6 hour print.  And the blobs stack on top of each other.  I do get clean walls though.  It's an option but if I could have the option to retract while combing, like for all other travels, that would be awesome.  It could be subject to the usual retraction settings, just an option to enable it while combing.  Please?  Honest it would be fantastic for PETG and TPU, they are so dribbly they really need this.

ctzsnooze, 2018-03-12:
Thank you!  If any testing of a build is required, let me know, happy to assist.

dmezh, 2018-03-13:
Having written #2531, I can also vouch for the immeasurable value of adding this setting. 

For extruder geometries that have a large hot zone, or don't have an actively cooled break (like the UMO, for example),

dmezh, 2018-03-13:
Sorry, last comment got cut off:

Having written #2531, I can also vouch for the immeasurable value of adding this setting.

For extruder geometries that have a large hot zone, or don't have an actively cooled break (like the UMO, for example), runny materials like PETG will leak during combing.

The essential principle for this change is exactly the same as ordinary combing. Sticky materials, like PETG, will build up on the nozzle and deposit a blob. Others may just lose pressure and string inside the object, negatively affecting the next printing move.

This is the exact. same. principle. Having the option to enable it during combing is valuable because it preserves the benefits and possibility of using combing with runnier materials and passively cooled hotends.

ctzsnooze, 2018-03-27:
Ian do you know the status of the retraction during combing queries?  I thought I saw somewhere that it would be introduced by Ghostkeeper, and that would have been fantastic news - now I am not sure?

Ghostkeeper, 2018-04-20:
There was an experiment more than a year ago on continuing retraction during travel moves, to keep an underpressure in the nozzle chamber. That never came to fruition though because it wasn't needed for our printers. It didn't make it any better.

ctzsnooze, 2018-04-21:
I am certain it would help with TPU when there are multiple combing travel moves along thing walls.

Without depressurising the nozzle, little bits ooze out the side of the nozzle during the combing.  When combing back and forth along long thin walls with large corner blocks, while using clear oozy filaments like clear PETG and TPU, yes there very definitely is a problem.  The walls get a sandpapery edge and look crap, not clear at all.  Pressure built up in the corners just oozes a tiny bit out the sides of the nozzle during the combing travels.  And if you have four corners there are LOTS of combing travels, it doesn't complete each corner and then just do the long perimeters, it runs around the object multiple times combing and spitting out tiny micro blobs all the time.

If I disable combing and instead travel over open space, the thin walls all lovely shiny and smooth again.  So I know for sure it is caused by combing.  But if I do this, I lose the benefit of combing, I get blobs that stick where the head crosses the edges.  

Please accept that there are users - quite a few - who are 100% certain that this problem would be fixed by providing a retract while combing option.  Could you please let us at least confirm the question using our materials on our printers?   If it doesn't work I'll be the first to apologise, but if it does, you'll make a lot of people very happy.

PS in your testing, you would not see any problem if combing over infill or large solid pieces with non-oozy filament.  i.e. I am sure if I tested with a big solid object there would seem to be no problem.  But it is a massive issue with TPU where there are multiple repeated combing moves along very thing walls like rectangular boxes.  It ruins the piece in this situation.

ctzsnooze, 2018-04-21:
PS this is not about retraction during travel, where retraction is well known to be vital with TPU.  It is about retraction during combing.  And for the very reason retraction during travel is important, there is every reason to expect that with TPU, and with objects like boxes with thin walls and lots of combing travels, that the same benefit from retraction would arise during combing as during other travels.

libbelibbe, 2018-04-29:
@adrianmihalko turn combing off. i've had the same issues.

ctzsnooze, 2018-04-29:
@adrianmihalko you are having leakage/dribbling despite retraction.  This is not the same problem I am raising in this issue.  In my case, with retraction on, everything is fine.  But because Cura does not permit retraction while coasting, I get this kind of leakage during the coasting moves, and that causes fine sandpaper effects on thin walls from un-relieved back-pressure leakage. Unfortunately Cura team don't appreciate that this is a real problem with clear runny materials like TPU and thin walls.

smartavionics, 2018-04-29:
> Unfortunately Cura team don't appreciate that this is a real problem with clear runny materials like TPU and thin walls.

Well, that's not entirely true - let's just say things take a while to change around here...

See https://github.com/Ultimaker/Cura/pull/3477

smartavionics, 2018-04-29:
I don't see why not, the default behaviour will be the same as it is now so it will not affect anyone who doesn't change the setting. I was not part of the previous testing that he refers to so I don't know what was done back then but just because it made no difference when tested on UM printers doesn't mean it isn't going to be useful for other printers. Don't give up hope (yet).

ctzsnooze, 2018-04-29:
Thanks Mark!  I very much appreciate your coding work here.   I do coding elsewhere on github, but because I can build the code immediately, I can test ideas out quickly.  This one is indeed taking a long time  :-)  Thin walls on my current TPU prints now come out OK with Z lift and no combing, but I get a bit of blobbing and a lot of very fine cotton wool like stranding, and the prints are slower with all the Z lifting, compared to combing.  So I really look forward to the combing fix, can't wait to getting back to using combing usefully.  Thanks again - C

Ghostkeeper, 2018-05-25:
The testing that we did here was not done with runny materials (we used PLA) but it was done with a very large nozzle size which was also runny when making large travel moves over air.

The thing that we tested was to keep retracting slowly while travelling. It had nothing to do with combing per sé. Let's hope that Smartavionics' solution works out better to reduce some of the underextrusion after oozing so much.

ctzsnooze, 2018-05-26:
Finger's crossed, very much looking forward to testing SmartAvionics's solution also.

g00bd0g, 2018-06-11:
Trying to print PETG parts with Cura is, well, trying... ;p

I have tried every possible combination of coasting/retracting/z-hop/scaleable priming, etc... to no avail. 

I would be extremely happy if the ability to coast and then retract before any/all travels would be enabled. I would even be happy to make a donation if that helps motivate someone :)

smartavionics, 2018-06-11:
Hello @g00bd0g, I print PETG with great success. I never use any combing and little or no fan.

g00bd0g, 2018-06-11:
Yes, I just discovered combing was messing me up! I am now a bit confused though, does retraction happen after coasting, before traveling?

smartavionics, 2018-06-11:
I don't use coasting but I would expect retraction to occur after coasting before the travel starts.

ctzsnooze, 2018-06-19:
@smartavionics - super excited!  I just got 3.4 beta (OK slow) and retraction with combing has made a huge improvement on my thin wall TPU prints!

Overall the results were massively better with retraction.  Now I don't need to use Z-retraction with its associated stringing problems, and get lovely smooth interior and exterior of the print.  Fantastic!

This is a really significant improvement.  

That's to everyone who helped make this possible!

Ghostkeeper, 2018-07-31:
I think we'll consider this issue solved by https://github.com/Ultimaker/Cura/pull/3477.

closed 2018-07-31 by Ghostkeeper