Ultimaker/Cura #3246: TypeError: type SettingFunction doesn't define __round__ method

goebish, 2018-02-07:
**Application Version**
3.2

**Platform**
Windows

**Steps to Reproduce**
Select Monoprice Mini as printer

**Actual Results**
App crash with TypeError: type SettingFunction doesn't define __round__ method in the log

I had no issue with 3.2 beta

Vandrasc, 2018-02-07:
Confirmed, thank you for reporting it! Devs see CURA-4923

fieldOfView, 2018-02-07:
See https://community.ultimaker.com/topic/21397-issue-with-cura-32-and-monoprice-select-mini-v2/?tab=comments#comment-199273

In the definition, value is set to a number (should be a python-interpretable string)

Appesteijn, 2018-02-07:
I searched through all printer definitions and in malyan_m200.def.json, peopoly_moai.def.json, renkforce_rf100.def.json a "value" is used instead of a "default_value" for "material_diameter".

jackha, 2018-02-07:
CURA-4928, working on it and it will be fixed in a bugfix release

fieldOfView, 2018-02-07:
The problem is not "value" or "default_value". The problem is that what comes after it should be a string in the case of "value" and a number in the case of "default_value".

correct:
"value": "1.75" 
or
"default_value": 1.75

incorrect:
"value": 1.75
or
"default_value": "1.75"

goebish, 2018-02-07:
Thanks, adding quotes to my custom definition file fixed the issue.

jackha, 2018-02-07:
We're changing it to default_value as most of the definitions use that. Thanks @fieldOfView for the valuable tip!

RichardSerton, 2018-02-07:
My 'definitions' folder is empty
[image]

Am I missing something here?

mherodev, 2018-02-10:
Is there a hotfix I can do, or should I download the beta?

Appesteijn, 2018-02-10:
We made a test version, it includes some bug fixes and a possible fix for the garbled texts. We have tested this version, but not extensively.
The download location is from my personal could storage, since we don't have another way to distribute 'daily' builds (yet).
Could you try Cura 3.2.1: https://stack.mybacklog.info/s/MgLnKXLL0P9ro4M ?

KRAUZAH, 2018-02-10:
Above fix worked for me. I downloaded Appesteijn's img and overwrote my 'Resource' folder with his and was able to add the printer without crashing. Haven't printed anything yet though.

rejas, 2018-02-11:
I can also confirm that the 3.2.1 test-version fixes the bug. In my case for the Renkforce RF100 printer who crashed with the official 3.2.0 version.

mherodev, 2018-02-12:
Aw, the file is no longer available.

Ghostkeeper, 2018-03-06:
Seems to be fixed now in these definitions.

closed 2018-03-06 by Ghostkeeper