Ultimaker/Cura #3247: 2nd PrintCore loaded with Breakaway heating up issue

grpiotr91, 2018-02-07:
Problems occurs when using breakaway material.
During the auto-leveling first extruder loaded with normal materials heats up normally (like 175/185*C). After the leveling first PrintCore, second one temperature is set only to 100*C, what causes (if there is only a little of non melted material) active leveling fail.

ianpaschal, 2018-02-26:
Hi, can you share your Cura logs or an example project causing this behavior?

closed 2018-03-09 by ianpaschal

Ghostkeeper, 2018-04-04:
This is a firmware issue. Cura doesn't control the active levelling procedure. I'll ask around.

Ghostkeeper, 2018-04-04:
I can't reproduce this though. The target temperature of both nozzles remains at 175C.

grpiotr91, 2018-04-04:
On mine machine this issue is still a problem, which firmware are you using?

grpiotr91, 2018-04-04:
[https://www.dropbox.com/s/t4zvba2wprsok4g/screenshot.JPG?dl=0](url)

Screenshot form Cura, standby temperature is set to 100, which is too low and sometimes causes automatic leveling fail.

Ghostkeeper, 2018-04-26:
I've been using UM3 firmware version 4.3.2.20180308 which is the latest stable version (back when I tested this and now still).

The stand-by temperature in Cura is not the temperature that it is (or should be) using for calibrating the active levelling. You can see the actual values that it's using by querying the API. You can query the API by visiting the webpage of your printer in your browser. There's a page there for the API and it includes a web-form that allows you to sample all printer data including the target temperature of the nozzles.