Ultimaker/Cura #3248: Crash in 3.2.0 when trying to add a Monoprice Select Mini printer

jonhendry, 2018-02-07:
**Application Version**

3.2.0

**Platform**

Mac OS X

**Display Driver**
NVIDIA 378.10.10.10.25.104

**Steps to Reproduce**
Bring up add printer UI. Select Monoprice Select Mini E3D. Click Add Printer button.

**Actual Results**
Application crashes. Sometimes the printer Machine Settings window flashes up in the split second before the app goes away.

**Expected results**
The Machine Settings window for the new Monoprice printer should be presented.

**Additional Information**
Stderr Logs messages:

2018-02-07 04:57:09,560 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-07 04:57:09,562 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-07 04:57:09,563 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 800, in event
2018-02-07 04:57:09,565 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 339, in event
2018-02-07 04:57:09,566 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
2018-02-07 04:57:09,568 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderStack.py", line 104, in <lambda>
2018-02-07 04:57:09,570 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 528, in updateMaterialForDiameter
2018-02-07 04:57:09,571 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]: TypeError: type SettingFunction doesn't define __round__ method
2018-02-07 04:57:09,573 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-07 04:57:09,575 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-07 04:57:09,576 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 800, in event
2018-02-07 04:57:09,578 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 339, in event
2018-02-07 04:57:09,579 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
2018-02-07 04:57:09,580 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderStack.py", line 104, in <lambda>
2018-02-07 04:57:09,582 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 528, in updateMaterialForDiameter
2018-02-07 04:57:09,583 - CRITICAL - [(140735883162432)-MainThread] cura.CrashHandler.__init__ [67]: TypeError: type SettingFunction doesn't define __round__ method

fieldOfView, 2018-02-07:
Duplicate of #3246

jackha, 2018-02-07:
CURA-4928, working on it and it will be fixed in a bugfix release

closed 2018-02-12 by nallath