Ultimaker/Cura #3250: Cura 3.2 incorrectly displaying text in sidebar on OSX

JustinW23, 2018-02-07:
**Application Version**
3.2 (also 3.2 beta)

**Platform**
OSX 10.13.4 on Macbook Pro Retina 15" Mid 2014

**Qt**
downloaded from Ultimaker website

**PyQt**
downloaded from Ultimaker website

**Display Driver**
NVIDIA GeForce GT 750M 2048 MB
Intel Iris Pro 1536 MB

**Steps to Reproduce**
Open the software.

**Actual Results**
The text in the right hand menu had white "blotches over the text"  Attached a screenshot. The whited out blotches scroll with the text

**Expected results**

**Additional Information**
Have tried complete reinstall. Also deleted the contents of ~\library\application support\cura\ folder. Tried a reinstall and same issue.  Reinstall 3.1 and the issue does not occur
[image]
[image]
[image]

Marlequeen, 2018-02-07:
Seconding this, I have exactly the same behavior on the exact same Mac book Pro release with OSX 10.13.2
I also tried the beta version, re-installing from scratch (also deleting all Cura folders) :

[image]

I have tried : changing language, changing theme to Ultimaker Dark, to no avail.
Cura 3.1 and 3.0.3, which I reinstalled, just work fine regarding this specific issue.

pbannister, 2018-02-08:
I also see this issue, on MacO3 10.13.3 with my MacBook Pro (Retina, 15-inch, Late 2013)

JustinW23, 2018-02-08:
I tried looking at those environment variables for QT but no luck at all.  It doesn't explain why  3.0 and 3.1 work fine though and 3.2 doesn't.

Appesteijn, 2018-02-09:
Devs see CURA-4941, https://github.com/Ultimaker/Cura/pull/3270

Appesteijn, 2018-02-09:
Could you try Cura 3.2.1: https://stack.mybacklog.info/s/MgLnKXLL0P9ro4M ? It is from my personal cloud storage, but currently we don't have a fast way of distributing a build (yet).

JackTirol, 2018-02-09:
Same problem here! Cura 3.2.1 didn't solve it. Went back to 3.1.0 which works fine.

OSX 10.13.3 on Macbook Pro Retina 15" Late 2013
[image]

JustinW23, 2018-02-10:
3.2.1 didn't fix the issue for me either.

Appesteijn, 2018-02-10:
I made a PR https://github.com/Ultimaker/Cura/pull/3279 in which all divisions and multiplications on Cura and Uranium are rounded. If this doesn't fix it, I give up.
On Monday I can create a new testbuild.

pbannister, 2018-02-21:
Yeh. The 3.2.1 version seems to be that same, with shredded text. Off to
figure out hold to build development versions from Github...

fieldOfView, 2018-02-21:
This thread has a new private build that should fix the issue:
https://github.com/Ultimaker/Cura/issues/3277#issuecomment-365911053

Alternatively you can install this plugin, which also fixes the issue:
http://files.fieldofview.com/temp/PrepareStageCompactor.curaplugin

ianpaschal, 2018-02-21:
@fieldOfView: Thanks again Aldo! Closing this thread since those links will solve the issue.

closed 2018-02-21 by ianpaschal