Ultimaker/Cura #3252: Tiny text on HiDPI display (Cura 3.2.0)

ungutknut, 2018-02-07:
**Application Version**
3.2.0 Appimage

**Platform**
Xubuntu 17.10

**Display Driver**
Intel HD 405, i915 standard driver

**Steps to Reproduce**
Start the program on a small Display with high pixel density (in my case 1920x1200 @ 7")

**Actual Results**
The text is very small. Next to unreadable.

**Expected results**
Readable text which uses the system DPI settings for scaling.

**Additional Information**
This seems to be a QT-problem in some way. I've set the system DPI setting to 180 which works great for all other programs. Only Cura doesn't seem to care. 

**What I've tried so far**
) deleting everything cura related in .config, .cache, .local/share
) setting "QT_AUTO_SCREEN_FACTOR=1" which has an effect (and worked sometimes in cura 2.7.0 Appimage [not reliably for some mysterious reason]) but not the one I was expecting:
factor=0: https://imgur.com/a/EC3pl
factor=2: https://imgur.com/a/LV7UT 
here's the super-enlarged splash-screen at factor 2 and some correctly scaled system font in the background for comparison: https://imgur.com/a/aEZLr
factor=1 basically gives the same result as 2
) running cura 2.7.0 from the stable-ppa: I remember that the font size with this version once was reasonable big. But the ppa-version stopped working from one update to the next. The latter is a known issue and the maintainer of the ppa stopped his work on it AFAIK; therefore I can't provide screenshots of the working state ( https://github.com/thopiekar/Cura-packaging/issues/31#issuecomment-351230105 )
) running cura 3.2.0 from the cura-master ppa: This version is crashing when trying to start, like 2.7.0 ppa.

thopiekar, 2018-02-07:
There is an environment variable for DPI scaling. Google for it. When set, it will tell the Qt runtime, to use this one instead.

ungutknut, 2018-02-07:
Oh, I guess I was mixing the env. variables up. It's not QT_AUTO_SCREEN_FACTOR which I'm supposed to set but rather QT_SCALE_FACTOR

A value of 1.5 gives really nice results:
http://i.imgur.com/6W3CwMC.png

Only the font looks a bit strange but that's ok.

Thank you very much for that hint!

closed 2018-02-07 by ungutknut

fieldOfView, 2018-02-07:
Does the font rendering improve when you specify a whole number (eg 2 instead of 1.5)?

thopiekar, 2018-02-07:
No problem. I know the pain, because I aready saw people having this
problem together with HiDPI displays. Often Cura is/was not fitting on the
screen, so I knew there is a working environment variable.

ungutknut, 2018-02-07:
> Does the font rendering improve when you specify a whole number (eg 2 instead of 1.5)?
Font looks good but the 3d-viewer eats up all the space:
http://i.imgur.com/Fm9qxo7.png

fieldOfView, 2018-02-07:
I find it interesting that not all text in your 1.5x screenshot is blurry, and intriguingly the text that is not blurry is the text that gets garbled in https://github.com/Ultimaker/Cura/issues/3250

ungutknut, 2018-02-07:
Not quite exactly. For me it's just the text that is changeable; the captions seem to render correctly.

fieldOfView, 2018-02-07:
The point that I was making is that the captions that render correctly for you are exactly the bits that render incorrectly for some others on OSX

ungutknut, 2018-02-07:
Oh, ok. That's right.

Ghostkeeper, 2018-03-02:
I saw a few commits coming by for Cura Connect that they were setting all their font sizes to be divisible by 2. Maybe that helps with font rendering on screens set to 150% pixel scaling?